jeelabs/esp-link #108: Transparent Bridge is Verry Slow

stockklauser, 2016-02-18:
Hello, 

i see the following problem: 

I have an RFID Reader and Send hin 3 Commands ( Packages with a max of 10 Bytes) and get a Answere of 9 bytes. 

With the Direct Conneciton i need around 35ms for this Command and over the ESP-Link Firmware 2.2 Beta 1 i need Around 200ms this is a factor or 6 times slower Transmission. 

Is this a Bug or what else?

stockklauser, 2016-02-18:
Short view of the DBG Messages

 '59799> Receive callback on conn 0x3fff49fc
 59799> Buf to UART 
 59810> Console Process 
 59810> TX 0x3fff49fc 9
 60014> Receive callback on conn 0x3fff49fc
 60014> Buf to UART 
 60025> Console Process 
 60025> TX 0x3fff49fc 9
 60229> Receive callback on conn 0x3fff49fc
 60229> Buf to UART 
 60240> Console Process 
 60240> TX 0x3fff49fc 9
 60446> Receive callback on conn 0x3fff49fc
 60446> Buf to UART 
 60456> Console Process 
 60457> TX 0x3fff49fc 9
 60661> Receive callback on conn 0x3fff49fc
 60661> Buf to UART 
 60672> Console Process 
 60672> TX 0x3fff49fc 9
 60875> Receive callback on conn 0x3fff49fc
 60876> Buf to UART 
 60886> Console Process 
 60886> TX 0x3fff49fc 9
 61092> Receive callback on conn 0x3fff49fc
 61092> Buf to UART 
 61103> Console Process 
 61103> TX 0x3fff49fc 9
 61309> Receive callback on conn 0x3fff49fc
 61309> Buf to UART 
 61320> Console Process 
 61320> TX 0x3fff49fc 9'

lucky01, 2016-04-28:
Any news about this ? Did you find a solution ?

tve, 2016-04-28:
> 60446> Receive callback on conn 0x3fff49fc
 60446> Buf to UART 
 60456> Console Process 
 60457> TX 0x3fff49fc 9

This indicates that esp-link transmits the 9 character response to an incoming message within 11ms. Given past experience, I suspect that the TCP timers on the host are at fault, specially if it's a windows box. One has to set TCP_NODELAY options and that can be tricky under windows. At least I've seen quite some trouble with that.

tve, 2016-12-11:
I don't know of anything I can do here. Please reopen if you have questions or there is an issue to fix.

closed 2016-12-11 by tve