jeelabs/esp-link #109: Supporting PXE?

shogsbro, 2016-02-23:
Would anyone else see value in supporting firmware updates via PXE/TFTP, as a method of applying software updates?

The firmware updates could apply to the esp8266 itself, or to an attached AVR.

Further reading:
- [Preboot Execution Environment](https://en.wikipedia.org/wiki/Preboot_Execution_Environment)

tve, 2016-02-27:
Mhh, interesting idea. But what would a workflow look like? Isn't it easier to use wiflash to push an update to esp-link? And to use avrflash/avrdude to push an update to the attached AVR?
If you wanted to implement PXE you'd have to find a way to get the DHCP proxy server for PXE, which probably means you need to implement a portion of the DHCP protocol 'cause the built-in one doesn't support PXE. And then you'd have to implement TFTP...

shogsbro, 2016-02-28:
I am looking for a way to script firmware updates against a database of MAC addresses vs firmware image desired.

[DNSMasq](http://www.thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html) provides the following:
> --pxe-service=[tag:<tag>,]<CSA>,<menu text>[,<basename>|<bootservicetype>][,<server address>|<server_name>]
Most uses of PXE boot-ROMS simply allow the PXE system to obtain an IP address and then download the file specified by dhcp-boot and execute it. However the PXE system is capable of more complex functions when supported by a suitable DHCP server.
This specifies a boot option which may appear in a PXE boot menu. <CSA> is client system type, only services of the correct type will appear in a menu. The known types are x86PC, PC98, IA64_EFI, Alpha, Arc_x86, Intel_Lean_Client, IA32_EFI, BC_EFI, Xscale_EFI and X86-64_EFI; an integer may be used for other types. The parameter after the menu text may be a file name, in which case dnsmasq acts as a boot server and directs the PXE client to download the file by TFTP, either from itself ( enable-tftp must be set for this to work) or another TFTP server if the final server address/name is given. Note that the "layer" suffix (normally ".0") is supplied by PXE, and should not be added to the basename. If an integer boot service type, rather than a basename is given, then the PXE client will search for a suitable boot service for that type on the network. This search may be done by broadcast, or direct to a server if its IP address/name is provided. If no boot service type or filename is provided (or a boot service type of 0 is specified) then the menu entry will abort the net boot procedure and continue booting from local media. The server address can be given as a domain name which is looked up in /etc/hosts. This name can be associated in /etc/hosts with multiple IP addresses, which are used round-robin.

shogsbro, 2016-02-28:
An example of [how to chain to a PHP script](https://wiki.fogproject.org/wiki/index.php/Advanced_Boot_Menu_Configuration_options)

shogsbro, 2016-02-28:
Source code for a PXE client, [iPXE](http://ipxe.org), [is on Github](https://github.com/ipxe/ipxe)

beikeland, 2016-10-15:
Not really offering a solution to PXE boot, but I fail to see what it offers that a regular http update would not offer?

http://esp8266.github.io/Arduino/versions/2.0.0/doc/ota_updates/ota_updates.html#http-server

You could easily pass MAC or other parameters to the update server and send the appropriate firmware without having to implement TFTP and limit yourself to a local broadcast network. Unless broadcast is the whole point of wanting PXE? Which I think you could more flexibly work around by having the update server broadcast a MDNS service the ESP could look for and issue a regular HTTP update towards?

tve, 2017-06-18:
closing 'cause nothing has moved here in a very long time.

closed 2017-06-18 by tve