jeelabs/esp-link #113: New esp-open-sdk release

apwiede, 2016-03-02:
- there seems to be a new release 1.5.2 of esp-open-sdk, I had to adapt the Makefile.

- there seem to be stronger syntax checks, I had to cast the 3rd parameter of getBoolArg:
    (bool *)(&flashConfig.......)

derSascha, 2016-03-28:
@apwiede thanks for the hint!

deece, 2016-03-31:
Casting to bool * is not advisable as bool is not guaranteed to be a single byte (and is typically an int).

Here's a different way to fix it.

deece, 2016-04-01:
Also relates to #75

bool used to be typedef'd to unsigned char (which was why no warnings were produced), but now comes from stdbool.h.

deece, 2016-04-01:
Some more errors (mostly format warnings coereced to errors):
mqtt/mqtt_cmd.c: In function 'cmdMqttDataCb':
mqtt/mqtt_cmd.c:45:3: error: format '%ld' expects argument of type 'long int', but argument 4 has type 'uint32_t' [-Werror=format=]
   DBG("MQTT: Data cb=%p topic=%s len=%ld\n", (void*)cb->dataCb, topic, data_len);
   ^
mqtt/mqtt_cmd.c: In function 'MQTTCMD_Subscribe':
mqtt/mqtt_cmd.c:171:3: error: format '%ld' expects argument of type 'long int', but argument 3 has type 'uint32_t' [-Werror=format=]
   DBG("MQTT: MQTTCMD_Subscribe topic=%s, qos=%ld\n", topic, qos);
   ^
mqtt/mqtt_cmd.c: In function 'MQTTCMD_Setup':
mqtt/mqtt_cmd.c:248:3: error: format '%lx' expects argument of type 'long unsigned int', but argument 2 has type 'uint32_t' [-Werror=format=]
   DBG("MQTT connectedCb=%lx\n", callback->connectedCb);
   
rest/rest.c: In function 'REST_Setup':
rest/rest.c:255:3: error: format '%ld' expects argument of type 'long int', but argument 4 has type 'uint32_t' [-Werror=format=]
   DBG_REST("REST: setup #%d host=%s port=%ld security=%ld\n", clientNum, rest_host, port, security);
   ^
rest/rest.c:255:3: error: format '%ld' expects argument of type 'long int', but argument 5 has type 'uint32_t' [-Werror=format=]
rest/rest.c: In function 'REST_Request':
rest/rest.c:351:3: error: format '%ld' expects argument of type 'long int', but argument 2 has type 'uint32_t' [-Werror=format=]
   DBG_REST(" #%ld", clientNum);
   ^
rest/rest.c:377:3: error: format '%ld' expects argument of type 'long int', but argument 2 has type 'uint32_t' [-Werror=format=]
   DBG_REST(" bodyLen=%ld", realLen);
   ^
rest/rest.c:396:3: error: format '%ld' expects argument of type 'long int', but argument 2 has type 'uint32_t' [-Werror=format=]
   DBG_REST(" totLen=%ld data=%p", headerLen + realLen, client->data);
   ^
rest/rest.c:415:5: error: format '%ld' expects argument of type 'long int', but argument 3 has type 'uint32_t' [-Werror=format=]
     DBG_REST("REST: Connect to ip %s:%ld\n",client->host, client->port);
     ^
rest/rest.c:423:5: error: format '%ld' expects argument of type 'long int', but argument 3 has type 'uint32_t' [-Werror=format=]
     DBG_REST("REST: Connect to host %s:%ld\n", client->host, client->port);
     
syslog/syslog.c: In function 'syslog_compose':
syslog/syslog.c:409:7: error: format '%lu' expects argument of type 'long unsigned int', but argument 3 has type 'uint32_t' [-Werror=format=]
       p += os_sprintf(p, ".%06luZ ", se->tick % 1000000);
       ^
syslog/syslog.c:416:5: error: format '%lu' expects argument of type 'long unsigned int', but argument 5 has type 'uint32_t' [-Werror=format=]
     p += os_sprintf(p, "%s %s %lu.%06lu %lu ", flashConfig.hostname, tag, se->tick / 1000000, se->tick % 1000000, syslog_msgid++);
     ^
syslog/syslog.c:416:5: error: format '%lu' expects argument of type 'long unsigned int', but argument 6 has type 'uint32_t' [-Werror=format=]
syslog/syslog.c:416:5: error: format '%lu' expects argument of type 'long unsigned int', but argument 7 has type 'uint32_t' [-Werror=format=]
syslog/syslog.c:418:5: error: format '%lu' expects argument of type 'long unsigned int', but argument 5 has type 'uint32_t' [-Werror=format=]
     p += os_sprintf(p, "%s %s - %lu ", flashConfig.hostname, tag, syslog_msgid++);
     
serial/console.c:79:3: error: format '%ld' expects argument of type 'long int', but argument 3 has type 'int32_t' [-Werror=format=]
   os_sprintf(buff, "{\"rate\": %ld}", flashConfig.baud_rate);
   
cmd/handlers.c:65:7: error: format '%lx' expects argument of type 'long unsigned int', but argument 3 has type 'uint32_t' [-Werror=format=]
       DBG("cmdAddCb: '%s'->0x%lx added at %d\n", callbacks[i].name, cb, i);
       
cmd/cmd.c:46:3: error: format '%ld' expects argument of type 'long int', but argument 3 has type 'uint32_t' [-Werror=format=]
   DBG("cmdResponse: cmd=%d val=%ld argc=%d\n", cmd, value, argc);
   ^
cmd/cmd.c: In function 'cmdParsePacket':
cmd/cmd.c:110:3: error: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'uint32_t' [-Werror=format=]
   DBG("cmdParsePacket: cmd=%d argc=%d value=%lu\n",
   
esp-link/cgiflash.c:33:3: error: format '%lX' expects argument of type 'long unsigned int', but argument 3 has type 'uint32_t' [-Werror=format=]
   os_printf("%p: %08lX %08lX %08lX %08lX\n", buf, buf32[0], buf32[1], buf32[2], buf32[3]);
   ^
esp-link/cgiflash.c:33:3: error: format '%lX' expects argument of type 'long unsigned int', but argument 4 has type 'uint32_t' [-Werror=format=]
esp-link/cgiflash.c:33:3: error: format '%lX' expects argument of type 'long unsigned int', but argument 5 has type 'uint32_t' [-Werror=format=]
esp-link/cgiflash.c:33:3: error: format '%lX' expects argument of type 'long unsigned int', but argument 6 has type 'uint32_t' [-Werror=format=]
esp-link/cgiservices.c:87:5: error: format '%lX' expects argument of type 'long unsigned int', but argument 7 has type 'uint32_t' [-Werror=format=]
     );
     ^
esp-link/cgiservices.c:87:5: error: format '%lX' expects argument of type 'long unsigned int', but argument 8 has type 'uint32_t' [-Werror=format=]
esp-link/cgiservices.c:87:5: error: format '%ld' expects argument of type 'long int', but argument 13 has type 'int32_t' [-Werror=format=]
[... 6 lines cut ...]
         ^
esp-link/cgioptiboot.c: In function 'processRecord':
[... cut ...]
   NOTICE("Flash map %s, manuf 0x%02lX chip 0x%04lX", flash_maps[system_get_flash_size_map()],
   ^
esp-link/main.c:152:3: error: format '%lX' expects argument of type 'long unsigned int', but argument 4 has type 'uint32_t' [-Werror=format=]

esp-link/cgipins.c:57:3: error: passing argument 3 of 'getBoolArg' from incompatible pointer type [-Werror]
   ok |= getBoolArg(connData, "swap", &swap);
   ^
esp-link/cgipins.c:58:3: error: passing argument 3 of 'getBoolArg' from incompatible pointer type [-Werror]
   ok |= getBoolArg(connData, "rxpup", &rxpup);

PR #134 (2016-04-03): Use update toolchain & fix errors discovered by it
The latest esp-open-sdk uses a more recent compiler with better quality checks.

It also avoids redefining C99 types defined in stdint.h & stdbool.h (eg. uint32_t).

These patches allow use of the updated SDK (and the included packaged Expressif SDK) by removing the stdint.h suppression (defining _STDINT_H), and fixing the warnings reported once the native types are used with the updated compiler.

This fixes #113 #75 #125
closed 2016-04-03 by tve