jeelabs/esp-link #117: Error when flashing Uno from Win7 pc

WheresWally, 2016-03-15:
Love this software. I have fairly easily managed to flash a ESP8266-01 and attach it to a simple UNO Arduino. the Arduino is counting and sending the count out the Serial.print.
I can see the data on the console.

Now I wanted to reprogram and I created a batch file and I get the error below.
Is my only option to use curl as I do not have a linux syste, to use?

```
avrdude: ser_open(): network connects are currently notimplemented for Win32 environments

avrdude done.  Thank you.

itsjustvenky, 2016-03-15:
@WheresWally  AVRDUDE network upload is not available on windows binary. It works only on Linux and Mac too (I guess).

minida28, 2016-03-15:
Try this one:
https://kuleuven.box.com/shared/static/ruwoo6q5kgmsiv93q2xp015iyq9fdwym.zip
is an unoffocial avrdude version compiled for windows with the net:hostname:port support
(tve posted this)

WheresWally, 2016-03-16:
That does run and attempt the program. BUT I get errors.
Thoughts?

The command I run
```
avrdude -Cavrdude.conf -D -V -patmega328p -carduino -Pnet:192.168.0.38:23 -b115200 -U flash:w:Counting.ino.hex:i -lerror.txt
```

The error
```
avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x33
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x37
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x34
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x0d
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x0a
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x0d
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x0a
avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude done.  Thank you.
```

And the LOG on ep-link
```
452908> Accept port 23, conn=0x3fff6f90, pool slot 0
453356> MCU Reset=gpio0 ISP=gpio-1
453451> serbridge: connection reset err=-9
851244> Accept port 23, conn=0x3fff6f90, pool slot 0
851542> MCU Reset=gpio0 ISP=gpio-1
852217> serbridge: connection reset err=-9
881233> 3fff2550 already freed
938491> Accept port 23, conn=0x3fff6f90, pool slot 0
938993> MCU Reset=gpio0 ISP=gpio-1
939090> serbridge: connection reset err=-9
```

WheresWally, 2016-03-16:
I have tried using CURL.
I wrote 1 batch files as below, and also there is the log file from the esp-link.

1.BAT
```
curl -m 1 -XPOST http://192.168.0.38/pgm/sync
```

2.BAT
```
curl -m 10 -XGET http://192.168.0.38/pgm/sync
```

3.BAT
```
curl -m 30 -XPOST -g -# -d Counting.ino.hex http://192.168.0.38/pgm/upload
```

Log File
```
728191> HTTP GET /log/text: 200, 60ms, h=20976
737453> UART 57600 baud
737454> OB init
737454> UART 57600 baud
737454> MCU reset gpio0
737464> HTTP POST /pgm/sync: 204, 11ms, h=21080
737569> HTTP GET /pgm/sync: 200, 55ms, h=20992
737881> UART 9600 baud
737881> MCU reset gpio0
738284> UART 57600 baud
738284> MCU reset gpio0
738687> UART 115200 baud
738688> MCU reset gpio0
739090> UART 57600 baud
739090> MCU reset gpio0
739487> HTTP GET /pgm/sync: 200, 59ms, h=20976
739493> UART 9600 baud
739493> MCU reset gpio0
739895> UART 57600 baud
739895> MCU reset gpio0
740298> UART 115200 baud
740298> MCU reset gpio0
740358> HTTP GET /pgm/sync: 200, 57ms, h=20976
740700> UART 57600 baud
740700> MCU reset gpio0
741103> OB sync abandoned after timeout, state=0 syncCnt=144
741103> UART 57600 baud
741103> OB init
741160> OB cannot sync
741215> HTTP GET /pgm/sync: 200, 59ms, h=20976
```

minida28, 2016-03-16:
Just making sure, have you wired the reset line from esp to arduino?
Default setting is from gpio12 on esp to the RST pin on the arduino.

minida28, 2016-03-16:
And below is the command I run. I use baudrate 57600 because I am using Nano.
For Uno I think baudrate 115200 (which you have used) is correct already.
```
F:\DRIVER\ARDUINO\Software\AVRDUDE\avrdude-windows-6.1-20140519>avrdude -DV -patmega328p -Pnet:192.168.10.173:23 -carduino -b57600 -U flash:w:UPLOADED_TO_POMPA_MQTT_ELCLIENT.ino.standard.hex
avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "UPLOADED_TO_POMPA_MQTT_ELCLIENT.ino.standard.hex"
avrdude: input file UPLOADED_TO_POMPA_MQTT_ELCLIENT.ino.standard.hex auto detected as Intel Hex
avrdude: writing flash (18406 bytes):

Writing | ################################################## | 100% 6.22s

avrdude: 18406 bytes of flash written

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.
```

WheresWally, 2016-03-16:
Yes, I can click on reset on the uC Console screen and the UNO resets.
I am using ESP8266 so it is GPIO0 that is hooked to RST.

minida28, 2016-03-16:
Hmm.. I am running out of idea mate...

WheresWally, 2016-03-16:
The Serial,begin is set to 57600 and before that was on 9600

WheresWally, 2016-03-16:
i will try 115200 tomorrow

minida28, 2016-03-16:
At what baudrate do you set on the uC console?
I think you set it to 115200

WheresWally, 2016-03-16:
I set the baud in uC Console the same as Serial,begin in the sketch that i loaded before i hooked up the esp-link,
I will try setting both to 115200.

Then i will try a much newer ESP8266-01 and then a ESP8266-12e

WheresWally, 2016-03-16:
SUCCESS
hey it worked, Set the baud in uC Console to 115200 regardless of the baud the sketch uses and it did work via the avrdude method.

Thanks

minida28, 2016-03-16:
Great then!

Cheers,

minida28, 2016-03-16:
The benefit of using avrflash is it will automatically select the correct baudrate whatever baudrate you set on the uC console

closed 2016-03-16 by WheresWally