jeelabs/esp-link #118: New Feature Request- Flash/Upload via the esp-link web page.

WheresWally, 2016-03-15:
Would it be possible to, on the left side menus,  have another option under uC Console called Program/Flash ?
On that Program/Flash page the user could provide the path to the HEX file and have it upload and flash from there.
Access to the Flash tab could also be password protected, so when you start to think about security later on.

tve, 2016-03-20:
This is certainly something someone could spend time on implementing. I have not found that it works terribly well with my workflows. I prefer to push updates to esp-link rather than use the UI to pull them from a web server I need to have running somewhere.

tve, 2017-06-18:
This have been implemented in v3

closed 2017-06-18 by tve