jeelabs/esp-link #119: REST/SLIP demo not working

WheresWally, 2016-03-17:
I have grabbed the EL-client and I am trying the REST example
I have checked Enable SLIP on the REST & MQTT page.

Any ideas?
The example fails like this

``` 
EL-Client sync failed!
�s�K�EL-Client sync failed!
�s�K�EL-Client starting!
�s�K�EL-Client sync failed!
�s�K�

marcov, 2016-03-17:
Hello,

For me SLIP+REST works after having fixed a bug in esp-link as explained here: #122.

I am not using directly el-client: I am doing a port of it in pure C as I don't use Arduino, you can find it here: https://github.com/marcov/c_elclient

With the most recent versions of esp-link the slip protocol has changed so maybe it does not match with the example you are using.

WheresWally, 2016-03-18:
I think I have version V2.1.7 ESP-link and version 2.2 of EL_Client,
I cant find the version of EL_Client that goes with V2,1,7

found this warning
```
The 2.2 series breaks SLIP (mqtt & rest) compatibility with 2.1.7: the standard slip control characters are used and all the REST and MQTT commands have been overhauled. Please also use the new el-client library on the arduino side.
```

So now I will try ESP_link 2.2 with the correct version of EL_Client,

minida28, 2016-03-18:
Hi,

If you use el-client, you should use esp-link version 2.2.x.
el-client will not work on esp-link version 2.1.x.

If you prefer to use the 2.1.x version, you should use espduino library instead of el-client.

cheers,

WheresWally, 2016-03-19:
Ok, so use the right versions and then it works. '

Thanks

closed 2016-03-19 by WheresWally

WheresWally, 2016-03-20:
The debug output should be at 9600 and be as below if the SD card is not inserted.When you insert the SD card the gold contacts are towards the front

ESP8266 Macro V1.01128kb SS_RX SizeCan't access SD card. Do not reformat.No card, wrong chip select pin, or SPI problem?SD errorCode: 0X1,0X0

Date: Fri, 18 Mar 2016 09:54:33 -0700