jeelabs/esp-link #120: ESP-LINK as time reference device

vbocci, 2016-03-17:
I am working in an educational scientific project for detection of Cosmic Ray .
The project is based on Arduino DUE and the device is connected to esp8266 running ESP-LINK.

http://arxiv.org/abs/1506.01915
http://arxiv.org/abs/1411.7814
http://www.slideshare.net/ValerioBocci/ardusipm-maker-faire-2015-rome

We use ESP-LINK as rs232-wifi in the basic way   just to collect data from our device. The last version of ESP-LINK beta2 has the MQTT and  NTP function and we want to use this feature to synchronize distant devices.

1) in wich way we can ask from arduino to esp-link   the time and to force an NTP request ?
2) I think it's a nice idea to provide a pin with a pps signal (like gps time receiver)  in this way ESP-LINK can be a time reference device at very low cost.

Thanks
Valerio Bocci

tve, 2016-03-20:
Currently esp-link uses the esp8266 SDK's NTP functionality, which sucks. It would be great if someone had the energy to implement SNTP from scratch so it can be invoked periodically. Maybe espressif adds such a feature to the SDK soon, though.

vbocci, 2016-03-21:
The only way I found to receive the correct hour from ESP-LINK is to grab the log at the startup. There is a way to force the NTP request and receive the time after the power up from arduino ?
I think to provide a pps is a relative easy job for people that know esp-link code.

Thanks
Valerio

ShaneKirkbride, 2016-06-06:
How accurate do you want it to be?

vbocci, 2016-06-07:
I think tens of ms  (< 50-100 ms ) can be an easy job,  few ms with a pps signal can be a final goal.

Valerio

tve, 2017-06-18:
closing 'cause nothing has moved here in a very long time.

closed 2017-06-18 by tve