jeelabs/esp-link #123: ESP_link V2.2 beta Not remote Program Arduino V1.7 did work ok

WheresWally, 2016-03-19:
So I know my hardware setup should be right, I was able to use arvdude to remote program an Arduino.
I am now using V2.2 so I could get SLIP functions, but now i cant get the remote flashing to work.
I run this
```
avrdude -Cavrdude.conf -D -V -patmega328p -carduino -Pnet:192.168.0.42:23 -b115200 -Uflash:w:rest.ino.hex:i 
```

and I get this ...
```
avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x45

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x4c
avrdude: stk500_initialize(): (a) protocol error, expect=0x14, resp=0x2d
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x43

avrdude done.  Thank you.

```

minida28, 2016-03-19:
Hi,

Normally I would check:

- SLIP is thicked on "REST/MQTT" screen at esp gui
- wire the reset from esp to uC reset pin, and make sure the uC can be reset using the 'Reset uC' button on the uC console screen at esp-gui
- you are using latest version of ep-link (v2.2.beta2 at the moment)
- baudrate at uC console should match the sync baudrate of the avr (115200 for uno I guess)
- try playing with RX-pullup thicked or un-thicked on 'Pin Assignment screen', at esp gui

cheers,

WheresWally, 2016-03-19:
so un-checking RX-pullup did allow it to program.
thanks

closed 2016-03-19 by WheresWally