jeelabs/esp-link #124: New Feature Request- ESP-link reset

WheresWally, 2016-03-19:
I need a way to reset the ESP8266 programmatically?

What would be nice feature is to have a second RESET button like theRESET on uC Console.

And also is there a command to RESET the ESP8266? or can I send AT+RST?

WheresWally, 2016-03-19:
Duh , I iound the reset ESP under the Debug Log tab,
cool

Still, there is esp.reset() in the demo.ino and that would be handy if it complied. Needs espduino.h

minida28, 2016-03-19:
Just my thought, I think you can use use a digital pin of your uC to pull down / LOW the reset pin of the esp.
But you will need a level shifter to do that - do not wire it directly.

tve, 2016-03-20:
I don't understand the point of having a programmatic reset call. If esp-link is functioning then why reset it and if it's not functioning why do you think the reset call will work?
In general, the way esp-link is intended is to control the life-cycle of the attached uC, not the other way around.
If you'd like to shut down esp-link from the attached uC then using the reset, or better, chip-enable pin is much better because (a) it guarantees that it actually shuts down or resets and (b) it saves power (which is a significant benefit).
I'm going to close the ticket, but if someone is inclined to implement a reset function I'm not opposed to it.

closed 2016-03-20 by tve

WheresWally, 2016-03-21:
The reason I wanted this was, after remote flashing the UNO with the rest.ino, the sketch would not sync, until I clicked reset ESP.
So I wanted to see if I could automate that.
When I design the PCB I will have a output connected to the ESP-RES pin.

I think of the 2 processors as peers, each checking the other and able to reset the other if it steps out of line.

minida28, 2016-03-21:
I suggest you use **avrflash**. The sketch will sync nicely again after flashing, without having to reset the esp. And also avrflash will detect the sync baudrate automatically.