jeelabs/esp-link #125: Problems with building on linux, Release 2.2.beta2, latest esp-open-sdk and ESP8266_RTOS_SDK_V1.4.0

TrystanLea, 2016-03-21:
I think I have followed the steps in installing the dependencies I have the latest esp-open-sdk (make STANDALONE=n), Not 100% absolutely sure about the SDK I have ESP8266 RTOS SDK v1.4.0. 
In the esp-link makefile I changed the paths too: (with my user)

XTENSA_TOOLS_ROOT ?= /home/user/Desktop/ESP8266/esp-open-sdk/xtensa-lx106-elf/bin/
SDK_BASE	?= /home/user/Desktop/ESP8266/ESP8266_RTOS_SDK_V1.4.0

With this I get:

VERSION: esp-link v2.2.beta2 - 2016-03-21 16:48:43 - 1d0f313
CC mqtt/mqtt.c
In file included from include/esp8266.h:5:0,
                 from mqtt/mqtt.c:40:
include/user_config.h:3:21: fatal error: c_types.h: No such file or directory
 #include <c_types.h>
                     ^
compilation terminated.
make: *** [build/mqtt/mqtt.o] Error 1

Trying esp-open-sdk/esp_iot_sdk_v1.5.2 that comes with esp-open-sdk nowI get a lot of errors of the form:

VERSION: esp-link master - 2016-03-21 17:42:39 - development
CC mqtt/mqtt.c
In file included from include/esp8266.h:5:0,
                 from mqtt/mqtt.c:40:
include/user_config.h:41:1: error: unknown type name 'uint8_t'
 extern uint8_t UTILS_StrToIP(const char* str, void *ip);
 ^
In file included from include/esp8266.h:14:0,
                 from mqtt/mqtt.c:40:
/home/trystan/Desktop/ESP8266/esp-open-sdk/esp_iot_sdk_v1.5.2/include/ets_sys.h:14:1: error: unknown type name 'uint32_t'
 typedef uint32_t ETSSignal;
 ^

and it goes on for a quite some time.

Can you see where Im going wrong?

Thanks a lot, Trystan

deece, 2016-03-31:
I worked around it by uncommenting the relevant lines in c_types.h in the SDK.

TrystanLea, 2016-04-29:
Thanks a lot, only just managed to test your recent updates now and it now compiles and uploads fine! really appreciated! Keen to try and use this as part of a heatpump monitor Im building here https://github.com/openenergymonitor/HeatpumpMonitor

tve, 2016-06-19:
I believe issues around using the SDK on non-standalone mode are fixed, but in any case, I highly recommend using standalone=y...
Pleas reopen if there are issues.

closed 2016-06-19 by tve