jeelabs/esp-link #126: Soft AP SSID minimum 8 characters?

jmcexx, 2016-03-21:
Using 2.2.beta2.  The configuration web page for Soft AP SSID does not allow the SSID to be less than 8 characters.  I'm guessing this is a bug in the web page code.

tve, 2016-03-22:
The Wifi standard requires SSIDs to have at least 8 characters IIRC.

martynj3, 2016-03-22:
Hi,

| 
  |  According to standard 802.11-2012 (Section 6.3.11.2.2), it can be 0-32 octets with an unspecified or UTF8 encoding, a zero length SSID is reserved for the wildcard SSID (in probe request frames for instance).
 - Martyn
 |

jmcexx, 2016-03-22:
I don't know about the standards, but all of my home routers allow short SSIDs.  Mine is only 3 characters right now.    

BTW, Thanks to you all for your work on this excellent project!

tve, 2016-03-22:
mhh, looks indeed like there are some checks in there that shouldn't be there and I didn't catch when reviewing. Passwords are also limited to `\w*` which prevents punctuation characters, not good. Thanks for flagging this!

commit a3772fc0 (2016-04-03): fix #126: allow short SSIDs
commit df2f36b8 (2016-04-03): fix #126: allow short SSIDs
closed 2016-04-03 by tve