jeelabs/esp-link #127: Flashing Esp-Link in Windows?

darkipod, 2016-03-21:
Im trying to flash the ESP-link files to my esp8266-01 in windows, Im currently writing all the files but after it does not give me anything on the status LED and I see no access point.

WheresWally, 2016-03-22:
You should be able to see debug data from the ESP8266 while it sets up. So after flashing, reset it, remove the FLASH jumper if that is how you get the ESP8266 to be in flash mode, and leave it hooked to the serial port of the computer, open a terminal at 115200 to that port, then reset the ESP8266.
Maybe you can post what you see there so the gurus can help you.

SloMusti, 2016-03-28:
Encountered the same problem with v2.1.7 when flashing using ESP tool on Windows on Nodemcu V2. UART debug on reset does not print out any logs, just some gibberish immediately after reset button is pressed.

The problem was in my case an old version of ESP Flasher tool, using the latest version from https://github.com/nodemcu/nodemcu-flasher solved the problem. @darkipod try this

After successful flash I get the serial output via USB.

tve, 2016-06-18:
Closing ticket: there's nothing actionable for me. Feel free to reopen.

closed 2016-06-18 by tve