jeelabs/esp-link #139: Problems building on Linux: esp_iot_sdk_v1.5.2 and newest version of esp-open-sdk

JonathanWitthoeft, 2016-04-12:
Makefile:73: Using SDK from /home/jonathan/esp8266/esp_iot_sdk_v1.5.2
VERSION: esp-link master - 2016-04-12 11:20:10 - development
CC mqtt/mqtt.c
CC mqtt/mqtt_msg.c
CC mqtt/pktbuf.c
CC mqtt/mqtt_cmd.c
CC rest/rest.c
CC syslog/syslog.c
CC espfs/espfs.c
CC httpd/httpdespfs.c
CC httpd/auth.c
CC httpd/base64.c
CC httpd/httpd.c
CC user/user_main.c
CC serial/crc16.c
CC serial/uart.c
CC serial/console.c
CC serial/slip.c
CC serial/serled.c
CC serial/serbridge.c
CC cmd/handlers.c
CC cmd/cmd.c
CC esp-link/cgimqtt.c
CC esp-link/mqtt_client.c
CC esp-link/cgiflash.c
CC esp-link/config.c
CC esp-link/log.c
CC esp-link/cgiservices.c
CC esp-link/cgioptiboot.c
CC esp-link/cgi.c
CC esp-link/task.c
CC esp-link/main.c
CC esp-link/cgipins.c
CC esp-link/status.c
CC esp-link/cgitcp.c
CC esp-link/cgiwifi.c
make[1]: Entering directory '/home/jonathan/esp8266/esp-link/espfs/mkespfsimage'
cc -I.. -std=gnu99 -DESPFS_GZIP   -c -o main.o main.c
main.c:26:18: fatal error: zlib.h: No such file or directory
compilation terminated.
<builtin>: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
make[1]: Leaving directory '/home/jonathan/esp8266/esp-link/espfs/mkespfsimage'
Makefile:477: recipe for target 'espfs/mkespfsimage/mkespfsimage' failed
make: *** [espfs/mkespfsimage/mkespfsimage] Error 2

tve, 2016-04-12:
looks like your system is missing zlib... apt-get install zlib or zlibdev or something like that? zlib is used to generate the read-only filesystem so you need it on the linux host, not on the esp8266 itself.

JonathanWitthoeft, 2016-04-12:
"apt-get install zlib1g-dev" did the trick. 
Thanks a lot. I am new to Linux. Sorry for a question unrelated to this project.

closed 2016-04-12 by JonathanWitthoeft