jeelabs/esp-link #142: Esp-link as a web server

cskarai, 2016-04-25:
Hi,

I'm playing with MCU-s to measure stuff and communicate data back to computer using esp-link.
The nicest solution is to create a web page which represents the data on a web interface.

esp-link has a pretty good httpd server, but I haven't found a solution to store web pages on esp-link instead of using the MCU.

If there's a 4MB module, 5x512k is the code, but 3MByte would remain to store web page for the MCU application. In this scenario MCU mesures, while esp-link handles the web interface.

What's missing from esp-link:
- OTA web page update (FTP or multipart HTTP POST)
- wire httpd to handle user pages as well

I'm open to implement similar patch on the code if it's accepted.

cskarai, 2016-04-25:
How I imagine this at the first glance:
- user can upload tar files onto the free partition > 1mb (sorry, but espfs is not common enough for uploading, this is why I chose tar, almost everyone can compress files into tar)
- implement tarfs (similar to espfs) for searching files in the uploaded tar archive 
- extend the web interface: 1st search in tarfs partition (> 1 mb), then in the normal espfs partition
- walk through over the folder list and catch (myapp/index.html) like file patterns, if this pattern is found, place myapp folder on the esp-link frame, so user can click on the application link onto it

I hope this solution won't consume much memory in the chip, but I think, that it could beneficial for the usability. It's also a question whether to implement this for <1Mb chips in the remaining section of the firmware.

tve, 2016-04-25:
These are good ideas. WRT file system, I'd be pretty resistant to add tar. Having two filesystems just adds unnecessarily to the codebase. There is everything in the makefile to produce an espfs and in order to compile esp-link anyone will have to get that to work anyway. Getting the YUI compression to work is actually much more messy and that doesn't change with tar. If you want to switch everything over to tar, I don't mind, but it has to work on linux and windows.

Before you implement, it would be good to discuss how the MCU sets the values that then get retrieved via json and how they're stored in esp-link... Is that a key=value linked list? Are all the values strings? Is that kept in RAM only?

cskarai, 2016-04-26:
Thanks for answering my question.

1. Let's keep espfs now and concentrate on the web server issue. If we move to tar, let's do it in another step

2. I'm not a web expert, so I may not have the necessary knowledge for architectural decisions.
- I intended to simply redirect pages ending with ".json" to the MCU (port 23) to handle the HTTP requests (not very nice, but requires minimal effort at the esp side)

I don't have much experience, but MQTT is also a solution, so MCU would publish data and an internal MQTT server would represent this on the web page (haven't used MQTT yet).

I don't have the exact answer for the question yet.

cskarai, 2016-04-28:
I've created a fork and a branch named "web_server" and I started working on that branch (just as a proof of concept, that it can work).

Some more details:
- esp-link should keep key-value pairs in the memory
- those keys will be embedded into HTML ( value="${key}" format )
- HTTP post on a form can change those internals
- MCU and esp-link should interact somehow (MQTT?)

I've started implementing the web-server part.

cskarai, 2016-04-29:
I've forked the project and started working on the implementation (no final code, just a sandbox).
cskarai fork -> web-server branch

The implementation is just a proof of concept solution, not intended for a pull request.

More details about the implementation:
- user can upload espfs file and httpd will handle that
- user html files will be shown on the left frame of the esp homepage
- esp-link can store a limited number of key-value-pairs in memory
- those values can be updated by either http post (html form) or MCU (with slip protocol)
- MCU to use slip protocol for key value pair communication

I'll inform you when the web-server prototype is finished.

cskarai, 2016-05-21:
Youtube video here:
[https://www.youtube.com/watch?v=d8byiqnbxRk](url)

minida28, 2016-06-24:
Nice work @cskarai! 

Any chance you can share your "sandbox" version ?  :smiley:

Hope this can be implemented into esp-link smoothly.

cskarai, 2016-09-04:
Created the pull request. It's a huge change.

It'll take some time to get in.

tve, 2016-09-23:
It's getting there :-)

closed 2016-09-23 by tve