jeelabs/esp-link #144: esp-link crash (or something like that)

thegimliboy, 2016-05-01:
I am using esp-link for a few month now, and it crashes every 6hours-1week... Then I have to manually restart it (the webinterface and telnet doesn't react anymore). No matter which version I use... I love it, but this is quite annoying.

Greets Gimli

tve, 2016-05-01:
Well, that's certainly annoying! I'm not seeing anything like that, so I'm not sure what causes the problem on your end. Have you looked at the debug log periodically and seen whether the heap size keeps decreasing? Have you been able to look at the debug log right after a crash (I assume it restarts) and captured the reset cause? Does it crash while there is traffic or just sitting idle? Are you just using it as a serial bridge or are you using the REST or MQTT functionality?

thegimliboy, 2016-05-01:
It crashes while sitting idle, and I use it as a serial bridge. No, I wasn't able to look at the debug log right after the crash. I have to unplug it to restart it, so the log was cleared after the restart.

tve, 2016-06-18:
Have you reproduced with 2.2.beta2? My esp-link's are running for weeks without any crash and I can't repro what you're seeing.

thegimliboy, 2016-06-19:
Oh. I used esp-link to get a wifi CUL, but while I was trying to switch my
esp to a new one, I realized, that my powersource wasn't working so well,
but it was too late and my CC1101 wasn't working anymore. I ordered a new
one, but still had no time to build a new setup. I will try to build it
again, and I will try the new version, but I guess it was my powersource.

tve, 2016-06-19:
Thanks! I'll close the ticket then. Please reopen if it happens again.

closed 2016-06-19 by tve