jeelabs/esp-link #146: Feature request: console page enhancements

guipoletto, 2016-05-18:
Hello, it would be possible to add a "Control+C", or "Ctrl+<whatever is in the command field" checkbox, along with carrier-return and newline boxes?

It is very usefull while using the ESP as a remote serial console into embedded-linux appliances. 

Also, allowing for the window to be resized (drag it diagonally in big screens) would make it nicer!

>>> Kind of a bug: for whatever reason, the telnet console always appears to send line-feed along with enter, resulting in double-shells for me. 

why does de-selecting the LF box in the web page does not affect the Telnet console?

tve, 2016-06-18:
WRT the ctrl-C, why are you not using a terminal or `nc` to connect to port 23? This way you can send all the control characters you want...

tve, 2016-06-19:
I cannot reproduce the issue with the LF box. It works fine for me:
```
$ od -c </dev/ttyUSB0
0000000   h   e   l   l   o   o   o   o   o   o   o   o   o   o   o   o
0000020   o   o   o   o   o   o   o   o   o   o   o   o   o   o   o   o
0000040   o   o   o   o   o  \r  \n   h   e   l   l   o   o   o   o   o
0000060   o   o   o   o   o   o   o   o   o   o   o   o   o   o   o   o
0000100   o   o   o   o   o   o   o   o   o   o   o   o  \r   h   e   l
```
Notice how the first helloooooooooooooooo... ends in cr-lf (both boxes ticked) and the second in cr on (LF unchecked).

tve, 2016-06-19:
Making the console resize would be a nice touch, but I'm not going to be able to get to this in the near future. PR welcome.
I'm going to close this ticket 'cause there's nothing really actionable for me. Please feel free to reopen.

closed 2016-06-19 by tve