jeelabs/esp-link #147: Longer MQTT server hostname

fedearne, 2016-05-19:
Please allow more than 31 characters in MQTT server hostname.

IBM Bluemix is free, but use rather long DNS aliases.

Thanks.
Morten

pcunha-lab, 2016-06-03:
I am looking for this fix too. Looking at the code at config.h, if i increase the size of mqtt_host or any other field will break all configurations after that address, correct?

If i proceed that way, all updates will loose their configurations.

 What shoud be the correct fix for that issue?

tve, 2016-06-03:
Yup, most likely the config version number needs to be bumped to avoid a mess, and ideally some migration code added that can take an old version and convert it by shifting up the bytes to make more room for the MQTT hostname.

commit dc4fabc7 (2016-06-19): Change MQTT hostname to 64 chars, fixes #147
closed 2016-06-19 by tve