jeelabs/esp-link #148: DHCP client can not connect

rudi48, 2016-05-28:
I flashed esp-link v2.2.beta2 - 2016-02-27 11:08:17 - 1d0f313 to an ESP-01 module.
Smartphone WLAN connect to the ESP Hotspot works.
Set Web browser to 192.168.4.1.
WLAN router was found, IP = 192.168.17.1
Unfortunately the ESP DHCP client could not connect to my WLAN router.
I have seen, that the ESP DHCP mask is 255.255.255.0.
What can I do to get a connect?
Regards, Rudolf

Please see the following debug log:
   311> connected with FRITZ-7170R, channel 11
   312> dhcp client start...
   312> Wifi connected to ssid FRITZ-7170R, ch 11

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x40100000, len 1456, room 16 
tail 0
chksum 0xa2
load 0x3ffe8000, len 776, room 8 
tail 0
chksum 0xc3
load 0x3ffe8310, len 552, room 8 
tail 0
chksum 0xb8
csum 0xb8

2nd boot version : 1.5
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 4Mbit(256KB+256KB)
jump to run user1 @ 1000

rl..rl....b.l.br|l.b.......l...!!!.
    93> 
    93> ** esp-link v2.2.beta2 - 2016-02-27 11:08:17 - 1d0f313
    93> Flash config restore *FAILED*
    94> CONN led=0
    97> Wifi init, mode=AP+STA
    99> Wifi uses DHCP, hostname=esp-link
   102> "ip": "0.0.0.0"
   105> "netmask": "0.0.0.0"
   107> "gateway": "0.0.0.0"
   110> "hostname": "esp-link"
   113> sleep enable,type: 2
   115> Serbridge pins: reset=12 isp=13 swap=0
   128> Reset cause: 0=normal
   129> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   129> Flash map 512KB:256/256, manuf 0xC8 chip 0x4013
   133> ** esp-link ready
   136> initializing MQTT
   138> MQTT_Init
   140> initializing user application
   143> Waiting for work to do...
   146> mode : sta(18:fe:34:9d:4b:7c) + softAP(1a:fe:34:9d:4b:7c)
   152> add if0
   153> add if1
   155> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   163> bcn 100
   284> scandone
   286> state: 0 -> 2 (b0)
   288> state: 2 -> 3 (0)
   299> state: 3 -> 5 (10)
   299> add 0
   299> aid 3
   299> cnt 
   317> 
   317> connected with FRITZ-7170R, channel 11
   317> dhcp client start...
   318> Wifi connected to ssid FRITZ-7170R, ch 11
this repeats forever ...

tve, 2016-06-18:
It crashes, and it's not clear why. Seems to block somewhere until it gets a watchdog reset. :-(
The only suspicious thing is "Flash config restore FAILED" which implies that it didn't properly save the flash config. It would be interesting to see what it logs when you change a setting in the web UI, for example the serial baud rate or such.
The only recommendation I can make it to erase flash (erase_flash command in esptool.py) and then flash all the pieces again. Sorry :-/

tve, 2016-07-04:
I don't think there's anything for me to do. Please reopen if there is or you have questions. I also recommend gitter.im for questions https://gitter.im/jeelabs/esp-link ...

closed 2016-07-04 by tve