jeelabs/esp-link #149: Setting mode to AP not remembered (#101 has returned for V2.2b2)

ArgoHG, 2016-05-31:
For 2.0 Beta 2 issue#101 is back. That is:
"After a reset, WiFi is back in STA+AP mode even if set to AP-only in web interface. Keeps running "Wifi checks" without SSID."

tve, 2016-06-19:
I cannot repro with master. I don't think anything has changed in this area. I switched from my normal STA mode to STA+AP, associated from a tablet and then switched to AP mode. I then reset using the button on the debug log page and it came back up in AP mode. I then power cycled and it again came back in AP mode.

tve, 2016-07-04:
I don't think there's anything for me to do. Please reopen if there is or you have questions. I also recommend gitter.im for questions https://gitter.im/jeelabs/esp-link ...

closed 2016-07-04 by tve

ArgoHG, 2016-07-05:
Hi Thorsten

Very sorry for the delayed reply, I've been traveling extensively (very
annoying when work gets in the way of hobby!). I used the code from here:
https://github.com/jeelabs/esp-link/releases/download/v2.2.beta2/esp-link-v2.2.beta2.tgz
and flashed it a number of times to a 12-F, with the same result. The fact
that it works for you is puzzling? Am I pulling the code from the correct
place?

At any rate by simply leaving the STA fields blank I get a good result and
it all works fine as an AP for my TCPclient.

Thank you again for developing and supporting this super useful code for
the 8266!

Warm regards,
Brendan (ArgoHG)