jeelabs/esp-link #151: Getting 962100 Serial Bridge Baud Rate?

matthewmellor, 2016-06-06:
Hi I'm just getting started using the esp-link firmware and I'm trying to send data through the serial bridge at 962100 baud. The web UI maxes at setting the value to 460800 baud but I figure there must be a setting that I could change within the code to raise the cap to 962100. Anyone know where I might start looking in the code? Thanks

tve, 2016-06-06:
If you look at the http request being issued by one of the baud rate change buttons you can simply change the number and issue it manually. Sorry, don't have it handy right this minute.
Note that you won't be able to sustain long bursts at 962100, there just isn't enough memory for buffering...

matthewmellor, 2016-06-08:
Thanks for your help. I was able to change what you said, managed to get it to compile, and am now stuck with the buffering issue. Do you know a work around for this? Buy a different esp with more memory? I'm using the feather huzzah esp8266 which uses the esp-12 and has 4MB of Flash. 
Alternatively, is there something I could do with the buffering? My end goal is to continuously stream data (from a force sensor which needs the high baud rate) to my computer and then to analyze it in matlab.

tve, 2016-06-18:
I'm not sure what to recommend. It's not going to be easy to stream at those data rates without packet loss. There are a bunch of tricks to save some memory that the esp-link code is currently using, but that will only get you so far.

I'm going to close this ticket 'cause there's nothing actionable for me. If you have further questions or issues please reopen.

closed 2016-06-18 by tve