jeelabs/esp-link #152: Best way to post data to http server?

ShaneKirkbride, 2016-06-06:
I would like to do a simple http post with this awesome library. What is the best way? I'd like to do something like press a button on a webpage and post some data to here: https://posttestserver.com for example. I saw there are some Ajax calls I could use...is this the best way?
Thanks!
-Shane

ShaneKirkbride, 2016-06-07:
Reading through the el-client code and the code here it looks like the best way to post data to a server would be via the uC. This would simply allow the ESP to be the "link".

ShaneKirkbride, 2016-06-16:
The trick is to get a uC with an open source SLIP interface...the STM32F4 does not have this.

tve, 2016-06-18:
El-client is indeed the code you need on the uC side. If you want something that is not arduino you're gonna have to adapt it. It's really not a lot of code, so it should be pretty easy.
I'm going to close this ticket 'cause there's nothing actionable for me. If you have further questions or issues please reopen.

closed 2016-06-18 by tve