jeelabs/esp-link #153: esp-01 crashes after boot

iot-werkstatt, 2016-06-13:
Hello guys,

I flashed esp-link v2.1.7 onto my esp8288 esp-01 module with the following parameters of esptool:
python esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash -fs 4m -ff 40m 0x00000 boot_v1.4\(b1\).bin 0x1000 user1.bin 0x7E000 blank.bin

The flash was successful and after restarting the module I can see the ESP SSID for a couple of seconds. But then the module crashes and the red LED goes off. Here is the serial output when connecting the module to Esplorer during boot until the crash happens:
   328> ** esp-link v2.1.7 - 2015-12-13 08:49:46 - cb75396
   328> Flash config restore *FAILED*
   329> CONN led=0
   330> Wifi init, mode=AP+STA
   333> Wifi uses DHCP, hostname=esp-link
   337> "ip": "0.0.0.0"
   339> "netmask": "0.0.0.0"
   342> "gateway": "0.0.0.0"
   344> "hostname": "esp-link"
   347> sleep enable,type: 2
   350> Httpd init, conn=0x3fff293c
   353> Serbridge pins: reset=12 isp=13 swap=0
   366> Reset cause: 6=external
   366> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   366> Flash map 512KB:256/256, manuf 0xE0 chip 0x4014
   371> ** esp-link ready
   373> MQTT_Init
   375> mode : sta(18:fe:34:f3:5e:16) + softAP(1a:fe:34:f3:5e:16)
   381> add if0
   382> add if1
   384> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   392> bcn 100
wUUUÕ¿ 11706> chg_A3:-180
 15349> Wifi check: mode=AP+STA status=0
 15350> Enabling/continuing uart log

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 1396, room 16 
tail 4
chksum 0x89
load 0x3ffe8000, len 776, room 4 
tail 4
chksum 0xe8
load 0x3ffe8308, len 540, room 4 
tail 8
chksum 0xc0
csum 0xc0

2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 4Mbit(256KB+256KB)
jump to run user1 @ 1000

don't use rtc mem data
f

Does anyone have an idea why this is happening? Maybe I did something wrong?

Thanks in advance....

Cheers

fab33, 2016-06-13:
It works fine wirth ESP-01 and esp-link 2.2.beta2

tve, 2016-06-18:
I'm not sure what to do with this. I'm not planning on fixing 2.1.7 if there is an issue. I'm going to close the ticket but please do reopen if there is additional info. I'll keep an eye out on whether I can repro.

closed 2016-06-18 by tve