jeelabs/esp-link #154: Configure Code to Use UDP Instead of TCP

matthewmellor, 2016-06-14:
Hi does esp-link support UDP in addition to TCP? If not would it be possible for me to configure the code to use UDP instead of TCP? Anyone have any ideas where I could get started?  I'm only interested in the serial bridge application so I can stream data. I've been using esp-link for about a week now using TCP and it's great (thank you so much) but I would like to try using UDP for my application instead. Thanks

tve, 2016-06-18:
The serial port is a byte stream protocol and TCP is also a byte stream. UDP is a datagram (packet) protocol, so you're gonna have some thinking to do...
The receive code for the serial brindge is at https://github.com/jeelabs/esp-link/blob/master/serial/serbridge.c#L157 and the transmit is at https://github.com/jeelabs/esp-link/blob/master/serial/serbridge.c#L241

I'm going to close this ticket (too many non-actionable tickets open) feel free to reopen if you have questions.

closed 2016-06-18 by tve

matthewmellor, 2016-06-18:
Thanks :+1: