jeelabs/esp-link #155: Steps to get started by a Newbie

matthewmellor, 2016-06-14:
I had some trouble compiling the firmware last week and finally got it to work so I wanted to share the steps I took. First though -> I would suggest using the released pre-compiled versions if you don't need to make any changes to the library. This pdf also have steps about how to go about flashing your esp in case you need it like I did. 

[BuildingEspLinkFirmware (3).pdf](https://github.com/jeelabs/esp-link/files/315072/BuildingEspLinkFirmware.3.pdf)

*Note these are the steps that I used to get the library to compile. I hope they work for you...though some of the steps may not be necessary (I'm still somewhat of a newbie at linux myself)

minida28, 2016-06-18:
Hi, I was also having a hard time compiling the esp-link so thanks a lot for this :+1: 

Yeah I think I agree that the key step is installing the esp-open-sdk with standalone mode i.e `standalone=y`.
If I use `standalone=n`, the compilations will always ended-up with errors.

Compiling esp-open-sdk runs fine, and it took about 50 minutes - it was on linux 16.04 running in virtual machine (1 processor and 2GB of RAM settings).

I need 2 additional steps to succesfully compile esp-link (in esp-link folder):

**Install java**
Using this step below will install Java default-jre version; other option is to install Oracle version.
```
$ sudo apt update
$ sudo apt install default-jre
```

**Edit the esp-link Makefile line 60**

From
```
SDK_VERS ?= esp_iot_sdk_v1.5.2
```
Into
```
SDK_VERS ?= ESP8266_NONOS_SDK_V1.5.4_16_05_20
```
Otherwise my compilation will stop with error below:
`make: *** No rule to make target '/ld/eagle.app.v6.new.1024.app1.ld', needed by 'build/eagle.esphttpd1.v6.ld'.  Stop.`

The line `ESP8266_NONOS_SDK_V1.5.4_16_05_20` was taken from the name of folder inside esp-open-sdk folder, depicted in the picture below:

[image]

Not sure, but I believe it got downloaded as part of esp-open-sdk compilations process. So maybe you must check and update it as necessary every-time you are using the latest esp-open-sdk.

Continue with flashing the newly compiled esp-link firmware with `./wiflash` and confirmed the new firmware from the esp-link GUI:

[image]

Hope this helps others with similar problems :)

blackadmin, 2016-07-08:
Seriously, This helped me a lot. I think the official docs are outdated. But still ESP-LINK is a very good work. :) 👍

ArgoHG, 2016-08-27:
So close but stopped by what I think is an out of date link :( Here is the error I get:

ArgoHG, 2016-08-27:
cd tools; wget https://htmlcompressor.googlecode.com/files/htmlcompressor-1.5.3.jar
--2016-08-27 12:12:37--  https://htmlcompressor.googlecode.com/files/htmlcompressor-1.5.3.jar
Resolving htmlcompressor.googlecode.com (htmlcompressor.googlecode.com)... 64.233.189.82, 2404:6800:4008:c01::52
Connecting to htmlcompressor.googlecode.com (htmlcompressor.googlecode.com)|64.233.189.82|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-08-27 12:12:37 ERROR 404: Not Found.

Seems like this refernce is put of date...?https://htmlcompressor.googlecode.com/files/htmlcompressor-1.5.3.jar

tve, 2016-08-27:
If you're trying to compile master you need v1.5.4.1, i.e. v1.5.4 plus patch 1

ArgoHG, 2016-08-27:
Hi tve.

Actually - I have success! I have the three .bin files and no errors.

To do this I edited the makefile here:

[122] GZIP_COMPRESSION ?= yes
   changing to: no
...
[129] COMPRESS_W_HTMLCOMPRESSOR ?= yes
   changing to: no

Probably a crude method but it seemed to get around my ERROR 404 problem, but what the hey.

ArgoHG, 2016-08-27:
I also had to edit makefile line 60 as per minida28's comment above. Plus the other tips posted above.

ArgoHG, 2016-08-27:
A small flashing tip: I use this guide and NodeMCU - http://www.electrodragon.com/flash-firmware-for-nodemcu-on-esp-12f-with-breakout-board/

ESPlorer (also in the linked guide) is a cool little program for testing serial comms as well.

And for testing the Port with a wifi TCPclient (I'm now using Port 23 or Port 4352) I use an Android phone with this app: "TCP/UDP TEST TOOL"

Anwaarullah, 2016-09-23:
I've setup the dev env on a Raspberry Pi and have uploaded the image here for anyone who'd like to build the esp-link firmware: [https://drive.google.com/file/d/0B-Z3b_6-UP8_U29QQmlrV1p6UEE/](https://drive.google.com/file/d/0B-Z3b_6-UP8_U29QQmlrV1p6UEE/)

The folders are setup in /opt/

**esp-link** contains the main branch and **esp-link-web-server** contains web-server branch by **cskarai**. 

Simply cd into any of these folders and **sudo make**

(It's an 8GB image file, so please flash the image to atleast an 8GB SD Card before booting on the Pi)

And also make sure the Makefile contained within the folders is edited for ESP flash size and other params.

Thank you @matthewmellor, @minida28 and of course @tve 👍

tve, 2016-09-23:
Thanks for the rPi image! I don't have any rPis to test, but I'm sure others will find this useful!

Anwaarullah, 2016-09-23:
To use the above RPi image (which was using SDK 1.5.4) against the latest master version, which is now using ESP SDK 2.0, please do the following:

cd /opt
wget http://s3.voneicken.com/esp_iot_sdk_v2.0.0.p1.tgx
sudo tar Jxf esp_iot_sdk_v2.0.0.p1.tgx
cd esp-link --> location where the latest master was cloned
sudo nano Makefile --> Make sure you customize params as per your ESP flash settings. Also make sure to change  COMPRESS_W_HTMLCOMPRESSOR ?= no 
sudo make

If your ESP is already connected to the network, simply 
sudo ESP_HOSTNAME=192.168.0.3 make wiflash --> to flash the latest firmware over to the ESP

afcarl, 2016-09-30:
Steps Followed to build esp-link v2.2.3 on a Win7 Pro x64:
1) Install Docker Toolbox ( http://www.docker.com/products/docker-toolbox )
2) Install Git Desktop ( https://desktop.github.com/ )
3) Clone esp-link from Github master to local repository ( https://github.com/jeelabs/esp-link )
4) Open Docker Quickstart Terminal
5) cd to local esp-link git repository ( C:\Users\xxxxx\Documents\GitHub\esp-link )
6) Run "docker run -v $PWD:/esp-link jeelabs/esp-link" command in  Docker Quickstart Terminal window

hetii, 2016-10-30:
Well today I try "docker" way to build firmware but when upload it to my ESP board (esp07) its end with resetting loop :/