jeelabs/esp-link #157: Not issuing Reset on program upload

Subhajitdas298, 2016-06-19:
I can reset using uC Console, but does not reset on avrdude program upload. Not even HTTP POST sync works. Says NOT READY.
But if I reset manually, program upload works fine.
I used user1.bin, ESP-12E module.
Need help.

tve, 2016-06-19:
The reset on upload is definitely issued. You can see the note in the debug log. Can you tell from your sketch whether it resets? It could also be a problem getting in sync after the reset...
In any case, without knowing which version of esp-link you're using and the debug log there's not much I can do.

Subhajitdas298, 2016-06-19:
Thank you for your attention. I have tried both versions. 
Here is log:
```

> 297> 
>    298> ** esp-link v2.2.beta2 - 2016-02-27 11:08:17 - 1d0f313
>    298> Flash config restore ok
>    298> CONN led=0
>    379> Wifi init, mode=STA
>    380> Wifi uses DHCP, hostname=esp-link
>    380> "ip": "0.0.0.0"
>    380> "netmask": "0.0.0.0"
>    382> "gateway": "0.0.0.0"
>    384> "hostname": "esp-link"
>    387> sleep enable,type: 2
>    390> Serbridge pins: reset=12 isp=14 swap=0
>    403> Reset cause: 4=restart
>    403> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
>    403> Flash map 4MB:512/512, manuf 0xE0 chip 0x4016
>    407> ** esp-link ready
>    410> initializing MQTT
>    412> MQTT_Init
>    414> initializing user application
>    417> Waiting for work to do...
>    420> mode : sta(5c:cf:7f:09:fd:61) + softAP(5e:cf:7f:09:fd:61)
>    426> add if0
>    428> add if1
>    429> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
>    438> bcn 100
>    439> f r0, scandone
>   2570> state: 0 -> 2 (b0)
>   2572> state: 2 -> 3 (0)
>   2576> state: 3 -> 5 (10)
>   2576> add 0
>   2576> aid 1
>   2577> cnt 
>   2588> 
>   2588> connected with SD Home, channel 9
>   2588> dhcp client start...
>   2589> Wifi connected to ssid SD Home, ch 9
>   3645> ip:192.168.1.201,mask:255.255.255.0,gw:192.168.1.1
>   3646> Wifi got ip:192.168.1.201,mask:255.255.255.0,gw:192.168.1.1

```

tve, 2016-06-19:
The log you pasted shows a successful programming: "OB pgm done: 200 -- Success. 1066 bytes at 115200 baud in 0.1s, 6704B/s 58% efficient"

Subhajitdas298, 2016-06-19:
That is what I am saying. Success, for first time, then later ones are errors. HTTP uploads works like charm. But I need this standard avrdude upload.

tve, 2016-06-19:
Mhh, I don't think I understand what you are doing. The log shows the built-in optiboot programming to work. So everything is wired up correctly and works, you just have to use the avrflash script, which is presumably what you did.
If using avrdude pointed at port 23 doesn't work, the problem is not one of resetting the AVR, the problem most likely is a sync issue. You would get info looking at the errors printed by avrdude, typically adding `-v` or multiple (`-v -v -v`) will get you the details. The reason I wrote the built-in optiboot programming code is that using avrdude was not reliable due to round-trip times. In many cases, the culprit was windows not flushing TCP buffers promptly. The AVR programming sequence has timing constraints and the avrdude sync code is not robust on top of that, so it becomes hit&miss...

Subhajitdas298, 2016-06-19:
This is not hit and miss, I tried with Virtual COM port on PC and avrdude directly on RPi (linux).
If I give long time rest, it is able to take new program. I think esp is not ending the session, or somehow dose not detect pattern, to reset for some time.
Here are some log, after a success and a faliure (I noticed faliure didn't change the log).

> f
> 232090> OB skip 2 cr/lf
> 232090> OB skip 2 cr/lf
> 232091> OB skip 2 cr/lf
> 232091> OB skip 2 cr/lf
> 232091> OB skip 2 cr/lf
> 232091> OB pgm 128@0x180
> 232108> OB skip 2 cr/lf
> 232108> OB skip 2 cr/lf
> 232108> OB skip 2 cr/lf
> 232108> OB skip 2 cr/lf
> 232109> OB skip 2 cr/lf
> 232109> OB skip 2 cr/lf
> 232109> OB skip 2 cr/lf
> 232109> OB skip 2 cr/lf
> 232109> OB pgm 128@0x200
> 232126> OB skip 2 cr/lf
> 232126> OB skip 2 cr/lf
> 232126> OB skip 2 cr/lf
> 232126> OB skip 2 cr/lf
> 232127> OB skip 2 cr/lf
> 232127> OB skip 2 cr/lf
> 232129> OB skip 2 cr/lf
> 232130> OB skip 2 cr/lf
> 232130> OB pgm 128@0x280
> 232147> OB skip 2 cr/lf
> 232147> OB skip 2 cr/lf
> 232147> OB skip 2 cr/lf
> 232147> OB skip 2 cr/lf
> 232147> OB skip 2 cr/lf
> 232147> OB skip 2 cr/lf
> 232148> OB skip 2 cr/lf
> 232148> OB skip 2 cr/lf
> 232148> OB pgm 128@0x300
> 232165> OB skip 2 cr/lf
> 232165> OB skip 2 cr/lf
> 232165> OB skip 2 cr/lf
> 232165> OB skip 2 cr/lf
> 232165> OB skip 2 cr/lf
> 232166> OB skip 2 cr/lf
> 232166> OB skip 2 cr/lf
> 232166> OB skip 2 cr/lf
> 232166> OB pgm 128@0x380
> 232183> OB skip 2 cr/lf
> 232183> OB skip 2 cr/lf
> 232183> OB skip 2 cr/lf
> 232183> OB skip 2 cr/lf
> 232183> OB skip 2 cr/lf
> 232184> OB EOF
> 232184> OB pgm 54@0x400
> 232194> UART 115200 baud
> 232194> OB init
> 232194> OB pgm done: 200 -- Success. 1078 bytes at 115200 baud in 0.1s, 6737B/s 58% efficient
> 103826> MCU Reset=gpio12 ISP=gpio14
.

And what do you mean that avrdude not reliable?
Sir, I think you should into your code listening to port 23.
HTTP is OK, as have to use avr-link for upload.

Subhajitdas298, 2016-06-19:
Here are some more logs. 
This is after 1 success and 2 fails:

> 2 cr/lf
> 232109> OB pgm 128@0x200
> 232126> OB skip 2 cr/lf
> 232126> OB skip 2 cr/lf
> 232126> OB skip 2 cr/lf
> 232126> OB skip 2 cr/lf
> 232127> OB skip 2 cr/lf
> 232127> OB skip 2 cr/lf
> 232129> OB skip 2 cr/lf
> 232130> OB skip 2 cr/lf
> 232130> OB pgm 128@0x280
> 232147> OB skip 2 cr/lf
> 232147> OB skip 2 cr/lf
> 232147> OB skip 2 cr/lf
> 232147> OB skip 2 cr/lf
> 232147> OB skip 2 cr/lf
> 232147> OB skip 2 cr/lf
> 232148> OB skip 2 cr/lf
> 232148> OB skip 2 cr/lf
> 232148> OB pgm 128@0x300
> 232165> OB skip 2 cr/lf
> 232165> OB skip 2 cr/lf
> 232165> OB skip 2 cr/lf
> 232165> OB skip 2 cr/lf
> 232165> OB skip 2 cr/lf
> 232166> OB skip 2 cr/lf
> 232166> OB skip 2 cr/lf
> 232166> OB skip 2 cr/lf
> 232166> OB pgm 128@0x380
> 232183> OB skip 2 cr/lf
> 232183> OB skip 2 cr/lf
> 232183> OB skip 2 cr/lf
> 232183> OB skip 2 cr/lf
> 232183> OB skip 2 cr/lf
> 232184> OB EOF
> 232184> OB pgm 54@0x400
> 232194> UART 115200 baud
> 232194> OB init
> 232194> OB pgm done: 200 -- Success. 1078 bytes at 115200 baud in 0.1s, 6737B/s 58% efficient
> 103826> MCU Reset=gpio12 ISP=gpio14
> 530839> Accept port 23, conn=0x3fff6f40, pool slot 0
> 837426> Accept port 23, conn=0x3fff6f40, pool slot 0
> 144013> Accept port 23, conn=0x3fff6f40, pool slot 0
> 376153> MCU Reset=gpio12 ISP=gpio14
> 685098> Accept port 23, conn=0x3fff6f40, pool slot 0
> 991890> Accept port 23, conn=0x3fff6f40, pool slot 0
> 258480> MCU Reset=gpio12 ISP=gpio14

Subhajitdas298, 2016-06-19:
Can I say that those MCU resets are after try? And what is the number in front (like 258480)?
So can I assume that it is listening to port 23 but can not reset my arduino (last 2 cases)?

tve, 2016-06-20:
Those resets are due to the avrdude connection. The number in front is ms since boot (wraps around at 1000000). It looks like the successful programming happens at 115200 baud. Is that what you have set in the uC console? The built-in optiboot programming auto-bauds but if you use avrdude it doesn't.

Here's what avrdude output looks like when it fails (I had the wrong baud rate):
```
/home/arduino/arduino-1.6.5/hardware/tools/avr/bin/avrdude -C /home/arduino/arduino-1.6.5/hardware/tools/avr/etc/avrdude.conf -DV -patmega328p -Pnet:esp-link8:2323 -carduino -b115200 -U flash:w:rest.hex:i
ioctl("TIOCMGET"): Inappropriate ioctl for device
ioctl("TIOCMGET"): Inappropriate ioctl for device
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x45
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x4c
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x2d
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x43
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x6c
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x69
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x65
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x6e
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x74
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x20
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude done.  Thank you.
```

And here's a successful attempt:
```
/home/arduino/arduino-1.6.5/hardware/tools/avr/bin/avrdude -C /home/arduino/arduino-1.6.5/hardware/tools/avr/etc/avrdude.conf -DV -patmega328p -Pnet:esp-link8:2323 -carduino -b115200 -U flash:w:rest.hex:i
ioctl("TIOCMGET"): Inappropriate ioctl for device
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: reading input file "rest.hex"
avrdude: writing flash (6524 bytes):

Writing | ################################################## | 100% 1.94s

avrdude: 6524 bytes of flash written

avrdude: safemode: Fuses OK (H:00, E:00, L:00)
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude done.  Thank you.
```

Several others and I have spent quite some time tweaking stuff in esp-link to try and make avrdude reliable and we concluded that it's just not possible due to the timing constraints. If it works: great, but if it doesn't it's very difficult to even troubleshoot and often there's not much one can do.

Subhajitdas298, 2016-06-20:
Thanks for elaboration. The output you showed is identical to my output, when flashing from linux. But those 'ioctl("TIOCMGET"): Inappropriate ioctl for device' do not happens for virtual port, in windows.

And yes it is set to 115200 baud, which is also used by arduino ide (as seen in boards.txt).

Subhajitdas298, 2016-07-03:
Can you tell me which part of your code is responsible for programming the arduino? So, I can look at it clearly. Please need to complete this project fast.

tve, 2016-07-03:
https://github.com/jeelabs/esp-link/blob/master/esp-link/cgioptiboot.c

tve, 2016-07-04:
I don't think there's anything for me to do. Please reopen if there is or you have questions. I also recommend gitter.im for questions https://gitter.im/jeelabs/esp-link ...

closed 2016-07-04 by tve