jeelabs/esp-link #164: ESP-LINK Flashes But NO Serial or Wifi AP

Applepi, 2016-06-27:
after flashing the ESP12 it doesn't respond to anything on the serial led when stuff is sent and there is no AP created, when flashing the blue led blinks and it finished with leaving, after than nothing.

Applepi, 2016-06-27:
br
 20256> 
 20256> ** esp-link v2.2.3 - 2016-06-21 21:58:48 - 1bcdc62
 20256> Flash config restore *FAILED*
 20257> CONN led=0
 20259> Wifi init, mode=AP
 20261> Wifi uses DHCP, hostname=esp-link
 20265> "ip": "0.0.0.0"
 20267> "netmask": "0.0.0.0"�

tve, 2016-06-28:
please provide the full log, not just a small snippet.

Applepi, 2016-06-28:
This is the full output, it doesn't continue past there and on boot it just
gives a blank console output that doesn't respond to serial without
flashing pins set

tve, 2016-07-03:
Maybe it's dying due to lack of power?

tve, 2016-07-04:
I don't think there's anything for me to do. Please reopen if there is or you have questions. I also recommend gitter.im for questions https://gitter.im/jeelabs/esp-link ...

closed 2016-07-04 by tve