jeelabs/esp-link #165: build fail <stdint.h>

vapor83, 2016-06-28:
can't get past this error, any suggestions?  Thanks

vapor@vapor-VirtualBox:~/esp8266/esp-link$ make
Makefile:73: Using SDK from /home/vapor/esp8266/esp_iot_sdk_v1.5.2
VERSION: esp-link master - 2016-06-27 23:49:27 - development
CC mqtt/mqtt.c
In file included from include/c_types.h:15:0,
                 from include/user_config.h:3,
                 from include/esp8266.h:8,
                 from mqtt/mqtt.c:40:
/opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                          ^
compilation terminated.
Makefile:500: recipe for target 'build/mqtt/mqtt.o' failed
make: *** [build/mqtt/mqtt.o] Error 1

tve, 2016-06-28:
If you're trying to build from master you need SDK 1.5.4

tve, 2016-07-04:
I don't think there's anything for me to do. Please reopen if there is.

closed 2016-07-04 by tve

rbenea, 2017-08-26:
I had a similar problem trying to recompile the lwip library.

```
cd ~/Arduino/tools/sdk/lwip/src
make

~/Arduino/tools/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                          ^
compilation terminated.
Makefile:20: recipe for target 'build/core/dhcp.o' failed 
```

The problem was that the compiler tried to include the standard stdint.h library located in the /usr/include folder. In order to work I had to modify the Makefile and -I/usr/include to the include list:

LWIP_INCLUDE = -Ibuild -I$(SDK_PATH)/include -I$(SDK_PATH)/lwip/include **-I/usr/include**