jeelabs/esp-link #167: Variable TCP PORT

rolfz, 2016-07-11:
Hi, Not an issue, but some applications require a fix port at an other value than 23 and are not flexible, so my question if it would be possible to add a port field that can be configured and stored in flash out of the WEB page ?? Would be great to have this featuer. (I am currently trying to connect my telescope with ESP-LINK and the IOS and Androit apps have fix port values).

yesyesuk, 2016-07-14:
I would also like very much if the port for the transparent bridge was configurable.

Ajlewka, 2016-09-01:
Need too

tve, 2016-09-01:
I one of you up to creating a pull request with the appropriate UI field?

mconsidine, 2016-10-06:
Hi,
I was wondering where this request stood?  What is needed for a pull request?
Thank you,

tve, 2016-10-06:
What's needed is making all the changes necessary end-to-end. Meaning: config variable to store the port, changes to use the port specified by the config variable, html element to display/choose the port, javascript to send changes, "cgi" handler to update the config variable.

mconsidine, 2016-10-06:
Got it.  Thank you.

fuzzball03, 2016-10-27:
I've submitted a pull request for this. Find it here:  pull #205
Spent an hour or so at work on this today...

I've only done really simple C coding, this will be my first that consists of multiple headers.
Please be easy on me - CONSTRUCTIVE CRITICISM WELCOME.

Note - this is untested.