jeelabs/esp-link #168: No network and fast flashing

vectorcrumb, 2016-07-13:
Using 2.2.beta2 on an ESP-01 blue, 4Mbit, the module will work correctly on the first reset after flashing. However, after removing power and powering on again, the module will repeatedly flash the status LED and no network will be created. When connecting via serial, the follow output flashes continuously.
[image]
Has anyone encountered this problem?

tve, 2016-07-13:
Mhh, that sounds strange. Do you have the same power source, 'cause not enough power will have this type of effect. You have the correct flash size config. I assume you did flash blank.bin to the correct location?

vectorcrumb, 2016-07-13:
Yeah, blank.bin is at 0x7E000 and user1.bin at 0x01000. The issue also appears when using 2.17 and 2.2.3. I'll try using a source with more power, but the issue has appeared up until now on a 1800 mAh battery and all USB3 ports on my computer (while plugged in).

leodesigner, 2016-07-19:
Looks like a bug in SDK.
Check this:
http://tech.scargill.net/esp-sdk-and-rboot-woes/

tve, 2016-12-11:
I don't know of anything I can do here :-(. Please reopen if there is more info.

closed 2016-12-11 by tve

igittigitt, 2017-01-15:
Currently i am faced with the same problem. Reboots sometimes does not work and now i only have fast blinking blue LED, no Wifi. I've read about issues when using espressif sdk 1.5.4, which is used in current esp-link compiling.
It should be much better in using SDK 1.5.3. Is it bad to use the older sdk? It would make the board more relaiable.