jeelabs/esp-link #170: Device Strange Behaviour

blackadmin, 2016-07-14:
After configuring STA
On Boot everything works fine, I can ping to device IP and can also open device web page in my browser.

Issue arises when I turn off my laptop wifi and then turn it on and then I cannot ping to ESP IP and I cannot even open ESP webpage, I dont know why, I tried accessing the ESP from my Mobile(NO WIFI RESTART) Browser still no response. 

And then after a while ESP can be pinged and ESP web page can be accessed (SAME IP) from my laptop and also from my mobile. 

I thought that ESP went offline, but ESP was connected to my cloud MQTT Broker when I was not able to ping the ESP.

blackadmin, 2016-07-15:
After some debugging, got why its happening.

This is due to some MQTT DNS thing, if I use MQTT hostname(domain) then after Wifi Router restart, MQTT is unable to connect to broker as mentioned in #169 and due to some MQTT DNS thing the above issue gets arises.

I tried using IP in MQTT broker, in that case ESP gets connected to MQTT Broker with no issue and also the above issue does not arises.

tve, 2016-07-15:
I'll look how DNS interacts with MQTT. Sounds a bit strange.

blackadmin, 2016-07-15:
Thank you very much :)

tve, 2016-12-11:
I'm not sure how to reproduce, I'm not seeing how DNS and MQTT would interact. Without more concrete steps to reproduce or esp-link debug printout there's not much I can do. Please reopen if you have more info.

closed 2016-12-11 by tve

Skoroglyad, 2020-08-18:
I've reproduced this issue. I used tls with 8883 port and I received "mqtt: dns lookup failed".  In the "Server hostname or IP" I set https://my-ip. 
After that, I used an HTTP connection with 1883 port and everything works fine.