jeelabs/esp-link #171: Problems with WiFi connections

Ema-tec, 2016-07-15:
I'm experiencing problems detecting and connecting to the on-board AP 
with both PCs and smartphones. The network seems invisible, and does 
not establish a link. I also tried the binaries found on this repo, with no positive results.

Any suggestions?

tve, 2016-07-18:
Mhh, this is a new one, haven't heard this before. How do you know that esp-link is running at all? Maybe posting the debug log from uart1 could help troubleshooting...

Ema-tec, 2016-07-18:
I know it is running because the AP appears and disappears on my devices and PCs. I also get 
output from serial, so the program did not crash. It seems that if there are too many, too few or 
too much powerful wifi signals (not related with the esp anyway), it tends to fail to create the AP. 
This issue stops both testing/debugging processes.

tve, 2016-07-20:
I'm sorry, I'm confused. Perhaps trying gitter would yield some clarity?
It sounds like you see the AP but it disappears before you can connect? I don't understand how "the network seems invisible" and "the AP appears [...] on my devices and PCs" can both be true at the same time...

Ema-tec, 2016-07-20:
I'm having both problems but at different times. It's either:
1) No network visible from my devices, even though the program is still flowing (I get serial output);
2) Visible network but no link can be established to the module. 
Sometimes on smartphones and/or PCs the softAP disappears, even if I'm 5-10cm apart from the module, and it's clearly not underpowered. Also small changes at times tend to compromise the workflow (even with a clean devel environment, commenting a variable and it's usage, or an ```if``` can mess things up), and I can't tell if it's the hardware being faulty or something else.

tve, 2016-10-23:
I'm not sure what I can do here. The esp-link code does not control the AP in any way, it's all in the SDK. Sorry.

closed 2016-10-23 by tve