jeelabs/esp-link #175: Unable to bridge Arduino & ESP Link

m960615, 2016-07-23:
First of all, this library is awesome..I am somehow managing to test things but now i really want to use mqtt with arduino  but i am unable to communicate. 

I am using esp-link v2.2.3 with latest elclient https://github.com/jeelabs/el-client 

Connected arduino uno rx & tx (pin 0, pin1) to esp. But when i check ESP Console i get below response 

```
15329> Wifi check: mode=AP+STA status=5
 15330> Wifi got IP. Going into STA mode..
 15330> bcn 0
 15330> del if1
 15331> pm~�ARDUINO: ESP not ready...
~�~�~�~�~�ARDUINO: ESP not ready...
~�~�~�~�~�ARDUINO: ESP not ready...
~�~�~�~~�ARDUINO: ESP not ready...
~�~�
```

Please help. Thank you.

tve, 2016-07-23:
Did you enable SLIP and MQTT in esp-link on the REST/MQTT page?

m960615, 2016-07-23:
Yes

tve, 2016-07-23:
Mhh, what you paste looks like the SLIP packets show up on the esp-link side and are not recognized as such, that's what all those odd chars are. Right now I'm struggling with what causes that.

m960615, 2016-07-28:
Is there anything i can do, to help you in debugging ?

tve, 2016-10-23:
The initial sync of el-client has been overhauled significantly. If you are still interested, please try again (you may want to wait for the next release). Please reopen if you are still having issues after that.

closed 2016-10-23 by tve

m960615, 2016-12-22:
The issue still exists till version  `esp-link v3.0.14-g963ffbb`, i am using latest el-client.

```
~�~�~�~�ARDUINO: ESP not ready...
~�~�~�~�~�ARDUINO: ESP not ready...
~�ARDUINO: starting
~U�~�~�~�~�~�ARDUINO: ESP not ready...
~�~�~�~ARDUINO: ESP not ready...
~�~�~�
```