jeelabs/esp-link #176: Unable to flash the firmware to ESP-07

crossband, 2016-07-29:
**Is there any guide how to flash the Esp-Link firmware to ESP-07?** 
I downloaded NodMCU Firmware Programmer, I tried flash with latest firmware, but I could not.
I would like to know about flashing addresses for each file and advanced parameters of NodeMCU programmer, e.g. Baudrate, Flash size, Flash speed and SPI Mode.
It was already installed some NodeMCU firmware before and it was working properly. Now I can not get it work. I really need it soon.

Please help.

tve, 2016-07-30:
It would be easier if you asked this type of question in gitter... There are also more people watching there.

If it really has 4MByte flash then the SPI mode should be QIO, just like the esp-12. The best thing to do is to erase flash and the reflash everything including the new esp_init_data_default.bin at 0x3FC000

ShaneKirkbride, 2016-08-04:
You can check this out: https://hackaday.io/project/10457-sunleaf/log/39299-elementary-esp-to-stm32-communication or ask the question in glitter...

Jufo84, 2016-08-11:
Hi,
Some progress in running on the ESP-07, very important to me. I will be grateful for your help.

NODEMCU report flashing is successful the ESP8266 reports Flash config restore failed on restarting. Not responding to AT or +++AT commands.

Wiring to flash ESP-07: http://dominicm.com/esp8266-flash-module-firmware/

Best Regards,
Robert

Jufo84, 2016-08-11:
It works!, try flash:
boot_v1.5.bin to 0x00000
blank.bin to 0x3FC000
user1.bin to 0x01000

ShaneKirkbride, 2016-08-13:
Can you say what you did to get it to work?

Jufo84, 2016-09-08:
Use these settings for flashing ( tested on 2 modules ESP-07):
boot_v1.5.bin to 0x00000
blank.bin to 0x3FC000
user1.bin to 0x01000

tve, 2016-09-23:
I'm glad you got it working!

closed 2016-09-23 by tve

saeidExe, 2021-06-13:
> 
> 
> It works!, try flash:
> boot_v1.5.bin to 0x00000
> blank.bin to 0x3FC000
> user1.bin to 0x01000

Hello dear friend . I have similar problem about update esp8266-07 (Blue Module) . What about esp_init_default.bin Address?

[image]