jeelabs/esp-link #179: Unable to flash connected uC (Arduino Nano) while receiving MQTT message about every 10 seconds or shorter

minida28, 2016-08-21:
Not sure if this is related to ElClient, Esplink, or problem in my sketch, but I cannot flash the connected uC (Arduino Nano in my case) while the uC was receiving mqtt message about every 10 seconds of shorter - I was using avrflash to flash eht uC.
Flashing process will start to success if I change the subscribing frequency to 15 seconds or longer. 

Below is error message when I run avrflash:
```
minida28@presario:~/arduino$ ./avrflash 192.168.10.173 pompa.hex
AVR in SYNC at 57600 baud: bootloader v1.16
Timeout waiting for flash page to be programmed
Error programming AVR
```
Sometimes the error goes like this:
```
minida28@presario:~/arduino$ ./avrflash 192.168.10.173 pompa.hex
Error checking sync: No response in state init(0) @57600 baud
```

To be able to flash the uC successfully, first I have to disconnect the esplink from the mqtt broker, then flash the uC firmware.

**_OR_**

Re-configure the mqtt subscribing frequency to about 15 seconds or longer.
```
minida28@presario:~/arduino$ ./avrflash 192.168.10.173 pompa.hex
AVR in SYNC at 57600 baud: bootloader v1.16
Success. 20714 bytes at 57600 baud in 5.5s, 3714B/s 64% efficient
Success, took 8 seconds
```
Well, this is not bother me too much when the uC is in operational. But sometimes we need to upload the firmware several times for testing, so any advise will be appreciated.

Thanks again for the great project!

tve, 2016-10-23:
I'm trying to understand this issue. Was the attached uC sending an MQTT message every 10 seconds, or was the broker sending a message every 10 seconds to the attached uC?
There currently is no lockout of the serial bridge while programming, so if some serial activity hits in the middle of the programming it will cause issues...

minida28, 2016-10-31:
hi @tve,
That is the broker sending a message every 10 seconds (or faster) to the attached uC will cause this issue.

If I slow it down a little bit to about 15 seconds or slower, I can say flashing success rate become 100% (i.e. always success).

commit 8fbb639b (2016-12-13): block MQTT during avrflash, fixes #179
closed 2016-12-13 by tve