jeelabs/esp-link #180: Multiple MQTT connection attempts

jakkubu, 2016-08-24:
I'm using firmware dependent on esp-link. During startup there were multiple mqtt connection attempts, despite the first one was successful. The problem was in `wifiStateChangeCb` function in [`mqtt_client.c line 69'](https://github.com/jeelabs/esp-link/blob/master/esp-link/mqtt_client.c#L69). MQTT was connected so the broker was disconnecting old connection and establishing new for each wifiStateChangeCb callback. Adding condition for `MQTT_CONNECTED` solved it:

        if  (  status == wifiGotIP &&
          mqttClient.connState != TCP_CONNECTING && 
          mqttClient.connState != MQTT_CONNECTED
        ) {

tve, 2016-08-24:
Makes sense. Thanks for the suggestion!

blackadmin, 2016-09-08:
@jakkubu  Thank you very much. 👍 :)

tve, 2016-10-23:
This is fixed on master. Thanks!

closed 2016-10-23 by tve