jeelabs/esp-link #182: Buffer problem

NJovanovicc, 2016-08-31:
Hello,
I am trying to use this great firmware ( esp-link -v2.2.3 ) for reading data from energy meter connected to WROOM-02 serial port.
I am sending  8 bytes read command  every second continuosly and get  17 bytes answer.
Works fine for a while but then answers stop,in the debug log:
**c tx buffer
310458> E:M 2928
310458> espbuffsend: cannot alloc tx buffer**
Any idea what is the problem and how to solve it?
Thanks,regards

tve, 2016-08-31:
The error says that malloc fails trying to allocate a transmit buffer of 2928 bytes. This happens inside the SDK code. This is a bit odd because there usually is quite some free memory. Esp-link periodically prints the amount of free memory, do you have the 20-30 last log lines leading up to the error?

NJovanovicc, 2016-09-01:
Thank you for your fast response.
Here it is:
**6164> ip:192.168.1.123,mask:255.255.255.0,gw:192.168.1.1
  6165> Wifi got ip:192.168.1.123,mask:255.255.255.0,gw:192.168.1.1
  6167> host_name = esp-link
  6169> server_name = _http._tcp.local
  8383> Thu Sep 01 07:30:27 2016
  8383> 
 14407> HTTP GET /pins: 200, 172ms, h=18440
 14407> HTTP GET /wifi/info: 200, 165ms, h=18632
 14507> HTTP GET /system/info: 200, 196ms, h=21032
 15205> Wifi check: mode=AP+STA status=5
 15206> Wifi got IP. Going into STA mode..
 15206> bcn 0
 15206> del if1
 15207> pm open,type:2 0
 15209> mode : sta(18:fe:34:db:9c:8b)
 15284> Turning OFF uart log
 16740> HTTP GET /log/text: 200, 9ms, h=20576
 16921> HTTP GET /log/dbg: 200, 206ms, h=22600
 30284> Wifi check: mode=STA status=5
 47757> Accept port 23, conn=0x3fff6510, pool slot 0
 53014> HTTP GET /log/text: 200, 239ms, h=22224
 93894> Accept port 23, conn=0x3fff66f8, pool slot 1
107854> HTTP GET /log/text: 200, 193ms, h=17272
162273> HTTP GET /log/text: 200, 156ms, h=22208
235946> Accept port 23, conn=0x3fff6850, pool slot 1
260247> HTTP GET /log/text: 200, 176ms, h=9752
348166> HTTP GET /log/text: 200, 146ms, h=21832

c tx buffer
611266> E:M 2928
611266> espbuffsend: cannot alloc tx buffer**
I forgot to mention that after some time I start getting answers again.

NJovanovicc, 2016-09-01:
Also, if it's any help, I just tryed the same thing with firmware "esp_iot_sdk_v1.5.2" and module working in 
UART-WiFi passthrought mode and everything is working fine.

tve, 2016-09-01:
You say using esp_iot_sdk_v1.5.2 as opposed to? What were you using before?

NJovanovicc, 2016-09-02:
As opposed to "esp-link -v2.2.3".

tve, 2016-10-23:
I cannot reproduce on master. Please let me know if you have specific steps for how to reproduce this.

closed 2016-10-23 by tve

NJovanovicc, 2016-10-26:
I tried testing it yesterday and everything worked fine for hours :) .
Please give me some more time to recreate the steps that led to this errror.