jeelabs/esp-link #184: Add more mdns txt records to support Arduino IDE flashing

beikeland, 2016-09-05:
In order to upload sketches from the newer versions of the IDE the device also have to broadcast  more txt records.

`upload_ssh=no` preferably `port=23`, there is also a parameter for `upload_auth=no|yes` but it does not seem to be required at this time.

Maybe you could add a check if the mDNS service name is _arduino._tcp you could also add the txt records to help the Arduino IDE?

I would have tried patching it, but I couldn't get the build env set up. esp-link is awesome btw!

tve, 2016-09-05:
thanks for the info! do you have a link to some doc or source where I can see the exact list of txt records?

beikeland, 2016-09-05:
Think this is the closest https://github.com/arduino/Arduino/pull/4107
I ran tests against 1.6.11 just before and its only upload_ssh that stops it from working now.

Just as a curious tidbit; oddly enough it was not successfull to forge the mdns records from another host, despite windows host seeing the records correctly, arduino ide ignored it.

elminister83, 2016-11-04:
the txt record are: ssh_upload not upload_ssh
this are the records from esp8266 arduino project
**tcp_check=no ssh_upload=no board=generic auth_upload=no**
and this from esplink
 **vendor\ =\ Espressif upload_ssh=no**
i can't make a pull request, excuse me
thanks

tve, 2016-11-04:
oops, thanks for the correction!

tve, 2016-12-11:
Looking at https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/packages/discoverers/NetworkDiscovery.java#L155-L168 it seems to me that tcp_check=yes is what we need and is default, and that auth_upload also defaults to no. I don't want to interfere with the board selection. So I think the only thing I need to do is to change upload_ssh to ssh_upload.

tve, 2017-06-18:
closing 'cause nothing has moved here in a very long time.

closed 2017-06-18 by tve