jeelabs/esp-link #186: Add UDP socket client, TCP socket client & server

beegee-tokyo, 2016-09-07:
I implemented UDP socket client, TCP socket client & server into esp-link & el-client.
Code is still under testing, but working.

Are you interested to add this functionality into esp-link/el-client? If yes, I can send a pull-request to start discussion.
Link to forks: 
[esp-link](https://github.com/beegee-tokyo/esp-link)
[el-client](https://github.com/beegee-tokyo/el-client)

mhanuel26, 2016-09-09:
Hi beegee-tokyo,

Do you think I might use your port as a starting point to add websockets?

Please let me know,

beegee-tokyo, 2016-09-09:
Yes, why not?

mhanuel26, 2016-09-11:
Hello beegee-tokyo,

I know you are not maintaining the esp-link but since you actually have contribute I would like to ask you or anybody that might give me some comments,

I am not seeing any documentation that this library has SSL/TLS support, not sure the reason since the inspired repo of tuanpmt/espduino supports it.

Do you know the difficulties or constrains to have secure sockets or mqtts with esp-link?

I will appreciate your comments,

rolfz, 2016-09-11:
Hello, Can I change the tcp/udp port on the web interface ? means it is not anymore limited to 23 and 2323 ?? and stored in eeprom ?
It was on the prinding enhancement list...

beegee-tokyo, 2016-09-11:
@rolfz: Ports 23 and 2323 are still used by used by the web interface. My UDP/TCP functions allow you to open a UDP socket or TCP socket connection to a remote server/client to transmit/receive data from the uC!. It has nothing to do with flashing a connected uC over the web interface.

You define the port you want to use when you call tcp.begin() or udp.begin().
Check the documentation [TCP begin](http://desire.giesecke.tk/docs/el-client/classELClientTcp.html#a7bf9b9d9375aabc9ad4a614d77c9016a) and [UDP begin](http://desire.giesecke.tk/docs/el-client/classELClientUdp.html#ad10f0990fa9cb4e75186b7570e37dcbe). You can see as well in my examples how to setup the port [TCP client example 1](https://github.com/beegee-tokyo/el-client/blob/master/ELClient/examples/tcp_client/tcp_client.ino) [TCP client example 2](https://github.com/beegee-tokyo/el-client/blob/master/ELClient/examples/tcp_client_resp/tcp_client_resp.ino) [TCP server example](https://github.com/beegee-tokyo/el-client/blob/master/ELClient/examples/tcp_server/tcp_server.ino) [UDP example](https://github.com/beegee-tokyo/el-client/blob/master/ELClient/examples/udp/udp.ino)

rolfz, 2016-09-11:
Thank you Beegee, I did not explain myself correctly, my question was about changing the port inside the web configuration page. I want to connect a wifi interface to my telescope to communicate with TCP on socket 1000 in state of 23 or 2323. I understand the original firmware only supports TCP/UDP over socket 23/2323. Would it be possible to change the port over the config page or do I need to recompile the firmware to change the port number ??

beegee-tokyo, 2016-09-11:
I am not working on anything to change the web-configuration pages, I only proposed to add some TCP/UDP socket connection functionality which can be controlled by the attached uC.
I remember your request from the issues list, but I don't think my code will solve your problems unless you are willing to use the new functions to setup the connection to the telescope from within the uC (Arduino board I guess) application to send/receive the data directly to/from the telescope.

mhanuel26, 2016-09-12:
Hello beegee-tokyo,

Would you mind to tell me the difficulties you foreseen make tcp/udp sockets or in general this project capable of SSL/TLS?

I will appreciate your response,

beegee-tokyo, 2016-09-12:
@mhanuel26: Sorry, I can not answer you. I didn't look (yet) into the ESP SDK regarding support of SSL/TLS. If there is native support, it might be not too difficult.

tve, 2016-09-23:
Closing since there's the pull request #188, please move discussions there. Thanks!!

closed 2016-09-23 by tve

Mauralva, 2017-02-03:
Can you help me a little more in the subject of create, connect, and comunicate between TCP Server and TCP Client?.
I have two NodeMCU DevKit with ESP-12. One of them its connected to a Arduino MEGA, and the other its connected to a Arduino Uno. Both of them are flashed with ESP-Link-v2.2.3
In the MEGA flash your [TCP Server Example] (https://github.com/beegee-tokyo/el-client/blob/master/ELClient/examples/tcp_server/tcp_server.ino) uncommenting the lines for de serial debug, and changing the ip of tcpServer and tcpPort
In the UNO I flash your [TCP Client Example] (https://github.com/beegee-tokyo/el-client/blob/master/ELClient/examples/tcp_client_resp/tcp_client_resp.ino), making the same change like in the other example.
The SLIC protocol seems to work fine.
The Debug of MEGA give me the fallowing message:
`192.168.4.100:7002 is served over connection number # = 0`
`EL-TCP ready`
and nothing more.
Te Debug of Uno give me the fallowing message:
`192.168.4.100:7002 is served over connection number # = 0`
`EL-TCP ready`
`Sending message to 192.168.4.100:7002`
repeating the last message indefinitely
By what I see in the code, the message sent is `"Message from Arduino Uno WiFi over TCP"`. But I can´t see that message in any where.

Can you help me what im doing wrong?

beegee-tokyo, 2017-02-03:
According to your comment, both ESP have the same IP. That is not possible. Give the server and the client different IP addresses and it should work.

Mauralva, 2017-02-03:
The ESP attached to the MEGA is configured as SOFT-AP with DCHP, and IP=192.168.4.1
The ESP attached to the UNO is configured as STA with DHCP, and  IP=192.168.4.2

The same ip that you see is the IP from server example.

`// IP address for this demo is a local IP.
// Replace it with the IP address where you have a TCP socket server running
char * const tcpServer PROGMEM = "192.168.4.1";
// Port for this demo is the port used by the TCP socket server.
// Replace it with the port that your TCP socket server is listening to
uint16_t const tcpPort PROGMEM = 7001;`

In the code, that variables are used in the fallowing statment
`tcpConnNum = tcp.begin(tcpServer, tcpPort, SOCKET_TCP_SERVER, tcpCb); // SOCKET_SERVER ==> accept connections`

I understand in that line I'm raising a server with that IP and Port. I'm in right?

I assume that the server TCP is in the AP, and the client is in the STA.

Then, in the client example I wrote the same ip and port of the server.
`// IP address for this demo is a local IP.
// Replace it with the IP address where you have a TCP socket server running
char * const tcpServer PROGMEM = "192.168.4.1";
// Port for this demo is the port used by the TCP socket server.
// Replace it with the port that your TCP socket server is listening to
uint16_t const tcpPort PROGMEM = 7001;`

And these variables are used in:
`tcpConnNum = tcp.begin(tcpServer, tcpPort, SOCKET_TCP_CLIENT_LISTEN, tcpCb); // SOCKET_CLIENT ==> we expect a response`

I understand in that line I'm connecting to one server in that IP and Port. I'm in right?

beegee-tokyo, 2017-02-03:
You're right, that should work. Misunderstood your first message.
Not sure why the server is not receiving the message. But I never tested the code with one ESP as AP and the other one connected to it. I have the server-client connection working with 4 ESP's, one is a client sending commands to the other 3 which are setup as servers. But the whole configuration is connected over my WiFi router/modem, not with one ESP as AP.
Can you test using both ESP's connected to an AP (which is none of the two ESP's)?

Mauralva, 2017-02-03:
I just tested like you recommend me (both esp conected to a Router. Then, one ESP configured as Server, and the other configured as a Client), but I have the same result.

Another question. If im connected to a network between the router, and i have the two ESP like i comment before, also i conect a PC to the network. Can I connect to the server created in one ESP in the IP 192.168.1.100 with thera term?

beegee-tokyo, 2017-02-04:
Beside of IP and port, did you change anything in the example code? I have no idea why it is not working for you.

I don't know Thera Term (Tera Term??). I use [PacketSender ](https://packetsender.com/) on Windows to test TCP/UDP servers and clients.

Mauralva, 2017-02-05:
Yes, I dont touch any more than the statments that I say before.

Yes, [Tera Term](https://ttssh2.osdn.jp/index.html.en) is the program. As I can see, Packet Sender allow you to create a Server/Client Socket. But Tera Term only act as client.

I am thinking the problem its in the Mega. I will use another kit next week and i tell you.

Many Thanks!!!

Mauralva, 2017-02-07:
Ok. I bought two new MEGA2560 and flash fresh firmware. Also connect the two ESP to a Router, acting both of them like STA. I put Serial1 to Serial, and Serial0 to debug (also try putting Serial0 as Serial and Debug)
The Output debug now show this:
In the server side:
```
EL-Client starting!                                            
ELC: got 10 @65C: 2 648 0 D9 2E                                
RESP_V: 1608                                                   
SYNC!                                                          
EL-Client synced!                                              
Waiting for WiFi ELC: got 14 @65C: 3 648 1 1 0 5 0 4A D9       
RESP_CB: 1608 1                                                
WIFI CONNECTED                                                 
ELC: got 10 @65C: 2 2 0 19 32                                  
RESP_V: 2                                                      
.2                                                             
                 
192.168.1.101:12345 is served over connection number # = 13340 
EL-TCP ready 
```
In the client side:
```
EL-Client starting!                                            
ELC: got 10 @6F3: 2 6C8 0 B7 3                                 
RESP_V: 1736                                                   
SYNC!                                                          
EL-Client synced!                                              
Waiting for WiFi ELC: got 14 @6F3: 3 6C8 1 1 0 5 0 A8 12       
RESP_CB: 1736 1                                                
WIFI CONNECTED                                                 
ELC: got 10 @6F3: 2 2 0 19 32                                  
RESP_V: 2                                                      
.2                                                             
                
Conectando a: 192.168.1.101:12345
TCP socket setup failed, try again in 10 seconds after reboot
```
I think the problem its in booth sides (client and server) becouse i use the Packet Sender to create a server o to send a packet to the server, and neither work.
Can you give me a guide to try make this work?

beegee-tokyo, 2017-02-08:
Beside of [example code](https://github.com/beegee-tokyo/el-client/tree/master/ELClient/examples) that you are using I have nothing I can share with you. 

What I saw is that you are using SOCKET_TCP_CLIENT_LISTEN in tcp.begin(). Just for testing change it on the client side.

On the client side try
`tcpConnNum = tcp.begin(tcpServer, tcpPort, SOCKET_TCP_CLIENT, tcpCb);`
instead of
`tcpConnNum = tcp.begin(tcpServer, tcpPort, SOCKET_TCP_CLIENT_LISTEN, tcpCb);`

SOCKET_TCP_CLIENT_LISTEN waits for a response from the server after sending a packet before it closes the connection. 
SOCKET_TCP_CLIENT just sends a packet and closes the connection.

Mauralva, 2017-02-21:
@beegee-tokyo finaly y could make it work. I dont know what was realy the problem. But after erase the esp memory with 1M.bin all in FF, and then flash the last Alpha version of esp-link, the API start to work fine.

Now I have another issue. In the API Doc, reference a class named [ELClientTcp](http://desire.giesecke.tk/docs/el-client/classELClientTcp.html) , but no .cpp or .h files with that names are in the git repo. I think the right class is named ELClientSocket.

Now Im decifrating the code becouse I need to make to do only few things: 2 o 3 modules bringing data to one that its acting as collector. Then, this burst all data collected to a PC when a certain command was received.

Thanks!

beegee-tokyo, 2017-02-21:
Good that you got it to work.

Documentation needs to be updated. You are right, ELClientTCP is no longer in the library, it is replaced by ELClientSocket.