jeelabs/esp-link #187: EL-Client sync failed!

iot-werkstatt, 2016-09-07:
Hello,

first of all I have to say that this firmware is fantastic. Great job!

However, I am facing an issue with the EL-client. I am using latest el-client library an Arduino Pro Mini 3.3V and version 2.2.3 of the firmware on esp01 1MB. MQTT status is enabled/connected, SLIP over serial is enabled, Serial baud is  115200. I see the MQTT status message in my MQTT network but in the microcontroller console is see:
EL-Client starting!
EL-Client sync failed!
ELC: got 9 @297: 103 F2054000 2085 34
ELC: Invalid CRC
EL-Client sync failed!
ELC: got 9 @297: 103 F2054000 2085 34
ELC: Invalid CRC
EL-Client sync failed!
ELC: got 9 @297: 103 F2054000 2085 34
ELC: Invalid CRC
EL-Client sync failed!
ELC: got 9 @297: 103 F2054000 2085 34
ELC: Invalid CRC
EL-Client sync failed!
ELC: got 9 @297: 103 F2054000 2085 34
ELC: Invalid CRC
EL-Client sync failed!
ELC: got 9 @297: 103 F2054000 2085 34
ELC: Invalid CRC

and the debug log shows:

 val=645 argc=1
230294> HTTP GET /console/text: 200, 207ms, h=20168
231305> HTTP GET /console/text: 200, 206ms, h=20168
231978> SLIP: start or end len=24 inpkt=0
231978> SLIP: start or end len=10 inpkt=1
231978> cmdParsePacket: cmd=1 argc=0 value=645
231979> cmdExec: Dispatching cmd=SYNC
231979> cmdResponse: cmd=2 val=645 argc=0
231979> cmdAddCb: 'wifiCb'->0x285 added at 0
231979> cmdWifiCb: wifiStatus=2
231979> cmdGetCbByName: cb wifiCb found at index 0
231979> cmdResponse: cmd=3 val=645 argc=1
233982> SLIP: start or end len=24 inpkt=0
233983> SLIP: start or end len=10 inpkt=1
233983> cmdParsePacket: cmd=1 argc=0 value=645
233983> cmdExec: Dispatching cmd=SYNC
233983> cmdResponse: cmd=2 val=645 argc=0
233983> cmdAddCb: 'wifiCb'->0x285 added at 0
233984> cmdWifiCb: wifiStatus=2
233984> cmdGetCbByName: cb wifiCb found at index 0
233984> cmdResponse: cmd=3 val=645 argc=1
235394> HTTP GET /console/text: 200, 303ms, h=20168
235987> SLIP: start or end len=24 inpkt=0
235987> SLIP: start or end len=10 inpkt=1
235987> cmdParsePacket: cmd=1 argc=0 value=645
235987> cmdExec: Dispatching cmd=SYNC
235988> cmdResponse: cmd=2 val=645 argc=0
235988> cmdAddCb: 'wifiCb'->0x285 added at 0
235988> cmdWifiCb: wifiStatus=2
235988> cmdGetCbByName: cb wifiCb found at index 0
235988> cmdResponse: cmd=3 val=645 argc=1
236316> HTTP GET /console/text: 200, 305ms, h=20168

Any idea what is going wrong here and how I can fix this?

Cheers

Daniel

iot-werkstatt, 2016-09-08:
**Update:** I tried the same setup with an Arduino Uno board and got it to work. I used the exact same sketch (mqtt example) from ELclient library. Does anyone have an idea why the Arduino Pro Mini with Atmega 328 3.3V 8M is not working OK regarding CRC?

beegee-tokyo, 2016-09-09:
Just a guess: 
The Uno is running with 16MHz, the Pro Mini 3.3V version runs only on 8MHz. Could it be that the Pro Mini has a receiver buffer overflow? Did you try with a slower baudrate?

iot-werkstatt, 2016-09-09:
Hi Bernd,

that is the solution. I reduced the baud rate to 38400 and it is working with the Arduino Pro mini now! Many thanks for this tipp! 

Cheers

closed 2016-09-09 by iot-werkstatt