jeelabs/esp-link #191: OTA ERROR USING UDOO: not in sync, inappropriate ioctl for device

stramike, 2016-09-20:
Hi all!!
I'm trying to upload a sketch on external arduino from udoo quad (ARM processor withUDOObuntu 2 OS based on ubuntu 14.04) using OTA upload by esp_link. When i give the shell command:

/home/udooer/Downloads/arduino-nightly/hardware/tools/avr/bin/avrdude -C/home/udooer/Downloads/arduino-nightly/hardware/tools/avr/etc/avrdude.conf -v -patmega328p   -carduino -Pnet:192.168.1.15:23 -b57600 -D -Uflash:w:/tmp/arduino_build_450821/rest_fix2.ino.hex:i

i get this output:

avrdude: Version 6.0.1, compiled on Apr  7 2016 at 22:44:28
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/home/udooer/Downloads/arduino-nightly/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/udooer/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : net:192.168.1.15:2323
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
ioctl("TIOCMGET"): Inappropriate ioctl for device
ioctl("TIOCMGET"): Inappropriate ioctl for device
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xc0
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x15
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x02
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x03
ioctl("TIOCMGET"): Inappropriate ioctl for device

avrdude done.  Thank you.

I'm using esp-link V2.2.3 on nodemcu(ESP-12), and using ARDUINO IDE 1.6.12 Hourly Build 2016/09/14 for ARM processor. I did point avrdude both on port 23 and port 2323 and doesn't work. I also switched -b option to 9600, 57600 and 115200 but i still get the same error.

What's going bad?
Thanks for any help

tve, 2016-09-20:
Your reset line may not be connected or you may be choosing the wrong pin on the esp8266. make sure you can reset the AVR using the reset uC button in the esp-link UI. It's also possible that you have the wrong baud rate: you need to set the correct one (looks like you want 57600) for programming in the esp-link UI.
You should also use the avrflash script, which uses the programmer built into esp-link, which is more reliable and provides better error messages. You can switch back to avrdude once you have the set-up working if you prefer that.

stramike, 2016-09-20:
Thanks for the quick reply!
I'm able to reset the AVR using the reset uC button in the esp-link UI. 
I double checked the baud rate (57600 on esp-link UI).
Then i used avrflash script by typing 

./avrflash 192.168.1.15 /home/udooer/Arduino/rest_fix2/rest_fix2.ino.hex

 and it gives me this output:

Error checking sync: sync abandoned after 8 attempts

I both tried to upload the hex file and the hex file with bootloader (I just created this two files using arduino IDE clicking Sketch --> Export compiled Binary and they appeared in the sketchbook folder).

tve, 2016-09-21:
Ugh, my bad, most of the debug output shows up in the esp-link debug console. You either don't have an optiboot compatible bootloader or there's something else mysterious in your circuit. Some people have had series resistors on the AVR TX -> esp RX line for "protection" that ended up causing issues.

stramike, 2016-09-26:
After power-cycle, upload esp-link firmware again, and do again all the connections all is working perfectly!! not sure what caused the issue.

Thanks a lot for the support!!

closed 2016-09-26 by stramike

swilson86, 2017-03-07:
TVE's comment on the avr-flash script rather than avrdude made my day. I spent WEEKS trying to fool with avrdude, serial uploaders, everything I could think of. avr-flash should be the standard that is written on the esp-link Home page under Info. I had it working in minutes.