jeelabs/esp-link #192: esp heats up at work

Jufo84, 2016-09-25:
Is it normal that the module esp-07 (esp-link) so much hot (+45 C). He works well with Roomba, but I ask whether it is normal :)

Best Regards,
Robert

tve, 2016-09-25:
this is not normal. at least I haven't found that they get this hot. are you sure it's the esp and not a voltage regulator you may have on the PCB that gets hot?
If it's the esp8266 then I assume that it's driving some output pin that has too small a pull-up resistor or that is connected to something that is also outputting causing the two drivers to "fight".

Jufo84, 2016-09-25:
Perhaps this voltage regulators on the board ESP-07, the whole is under the metal sheld. The external voltage regulator step down is set to 3V. Here are the photos of my module:
http://icebox.pl/roomba/roomba_01.jpg
http://icebox.pl/roomba/roomba_04.jpg
http://icebox.pl/roomba/roomba_02.jpg
http://icebox.pl/roomba/roomba_03.jpg

Module praucje praiwdłowo, komuniakcją the Roomba acts, the fact that the Roomba always sending data.

tve, 2016-09-25:
Nice PCB! There is no voltage regulator under the metal shield, just he esp8266 and the flash chip plus some passives. You are feeding it with 3.3v not 5v, right? If you can, measure the power current, it should consume around 70-80mA.

tve, 2016-09-25:
> I did not use resistor pull-up / down (10k?) for gpio15 and gpio 0, gpio 2 is not connected.
Most likely that's your problem. gpio2 needs to be pulled high to ensure proper booting. Others have reported issues when gpio15 is hard-wired to gnd. The chip outputs its clock at reset on gpio0, so hard-wiring that is not a good idea either. If esp-link is configured to output one of the LED signals on gpio0 or 15 you're also going to run into trouble...

Jufo84, 2016-09-25:
>Nice PCB! There is no voltage regulator under the metal shield, just he esp8266 and the flash chip plus some passives. You are feeding it with 3.3v not 5v, right? If you can, measure the power current, it should consume around 70-80mA.

THX :) Yes are feeding it with 3V (for safety), current 85mA, peak to 110mA.

> Most likely that's your problem. gpio2 needs to be pulled high to ensure proper booting. Others have reported issues when gpio15 is hard-wired to gnd.
Ok, gpio15 is hard-wired to gnd and gpio 2 is not connected, I should let go both (gpio 2 and gpio 15) to the ground by a resistor on what value?

tve, 2016-09-25:
The proper way to connect the esp8266 is to use a pull-up resistor in the 4.7k-10k range for gpio0 and gpio2, and a pull-down for gpio15. You need to drive gpio0 low at boot to get into flash-programming mode but not otherwise.
85mA avg sounds about right. You should really use 3.3v or the wifi might not work as well, e.g. less power. If the metal case gets warm then I would say it's OK, if it gets too hot for your finger then you have a problem as far as my experience goes.

Jufo84, 2016-09-25:
> You should really use 3.3v or the wifi might not work as well, e.g. less power. 

According to the manufacturer ESP-07 (datasheet) should work properly from 3V to 3.6V.
I set 3V because the 3.3V to gpio 12 (from the wakeup) happened cumulatively in peak 3.6V. When set to 3V, for gpio 12 is the peak 3,25V. It is based on the proven theory schema, which I modified my version.

> You need to drive gpio0 low at boot to get into flash-programming mode but not otherwise.

Yes, of course I have two pins (GPIO 0 and GND) and jumper for flashing

> If the metal case gets warm then I would say it's OK, if it gets too hot for your finger then you have a problem as far as my experience goes.

I measured the meter was maximally ~ +45 C, it was never too hot for my finger :)

Jufo84, 2016-09-26:
I researched myself ESP-07 without a voltage regulator, directly handed 3.3V. Without connecting TX and RX heats up to + 38C, something must be wrong, tomorrow will use a pull-down 10k for GPIO15.
GPIO2 is disconnected - there is a need to do pull-up? GPIO0 are disconnected. 
I looked at different schemes to connect, whether for CH_PD need to use pull-up? (I have connected directly to Vcc).

tve, 2016-09-26:
you should definitely pull-up gpio0 and gpio2. If you don't it may work, but can be sensitive to noise. You can hard-wire ch_pd unless you want to drive it from gpio16 for wake-up from deep sleep.

Jufo84, 2016-09-26:
> you should definitely pull-up gpio0 and gpio2

But gpio0 is connected only when flashing, I understand that in the normal mode, the boot is needed Connection pull up. Why pull-up for GPIO2?
Schematic for the ESP-12: http://souliss.net/images/2015-07/ESP_min.png

tve, 2016-09-26:
To get to the source, download the esp8266 pin list http://www.espressif.com/en/support/download/documents?keys=&field_type_tid%5B%5D=14 and look at the strapping tab. gpio0, gpio2, gpio15 form a 3-bit field that determines where the chip boots from.

Jufo84, 2016-09-26:
Thank you, summing:

Normal boot: 
GPIO15 (MTDO), pull-down (L)
GPIO0, pull-up (H)
GPIO02, pull-up (H)
CH_PD (H) - Optional!

Flash boot:
GPIO15 (MTDO), pull down (L)
GPIO0, pull-down (L) - connect to GND
GPIO02, pull-up (H)
CH-PD (H) - Optional!

BGatRMIT, 2016-09-26:
G'day Jufo84,
Just a rule of thumb following on from TVE above. Any pin set to be an
input on a micro should be pulled high or low. (Depending on what you want
it's default state to be).
Otherwise the input will "float" and randomly be either state. Just the
capacitance from your hand waving near it can change the state.
So by tying it to one state or the other via a resistor ensures predictable
behaviour.

Ben

JohnOH, 2016-09-27:
To keep battery consumption at a minimum and assuming no other constraints, choice of high or low depends on the typical state of the pin in actual operation.

John.

Jufo84, 2016-09-27:
I made a modify the PCB, I added a pull-up / down resistor 10k, and continues to heat up :(
http://icebox.pl/roomba/mod.jpg

Jufo84, 2016-09-28:
My ESP-07 died after 2 hours after modifications pull-up / pull-down :( I decided to buy tested ESP-12. It good wirinig ?, but gpio2 everywhere is not used:
http://www.electrodragon.com/w/images/8/86/Wi07_smd_wiring.jpg
https://ae01.alicdn.com/kf/HTB1a.JWKpXXXXc_XVXXq6xXFXXX4/201898713/HTB1a.JWKpXXXXc_XVXXq6xXFXXX4.jpg

BGatRMIT, 2016-09-28:
G'day Jufo84,
That's unfortunate. Sounds like the 07 was faulty.
Not quite sure what you are asking though re the 012?

Ben

Jufo84, 2016-09-28:
I would like to confirm that the wiring diagram ESP-012, who presented is correct. ESP-12 is more popular and officially supported by the esp-link.

BGatRMIT, 2016-09-28:
Well I have run one without a problem for over a month now. Occasional
serial glitches but I suspect that's the arduino not the esp.
The main fault I have researched with the ESP is the stability of the psu.
Ie if you don't provide enough mA and caps to keep the Voltage stable they
reset themselves.

Ben

commit 68fd831f (2016-10-23): document flashing HW, closes #192
closed 2016-10-23 by tve