jeelabs/esp-link #194: Does sep-link compiled with SDK_2.0.0.p1 fit in .iram ?

katast, 2016-09-29:
I have tried to compile esp-link with sdks 2.0.0.p1 but I always get the error '.text' will not fit in region 'iram1_0_seg'.

I have noticed a huge increment in .iram usage in the new sdk. Does it fit in the ram when you try to compile it?

I had a look to the Makefile and I haven't seen in the flash option any reference to esp_init_data_default.bin and there is boot_v1.5.bin instead of boot_v1.6.bin.

`flash: all
	$(Q) $(ESPTOOL) --port $(ESPPORT) --baud $(ESPBAUD) write_flash -fs $(ET_FS) -ff $(ET_FF) \
	  0x00000 "$(SDK_BASE)/bin/boot_v1.5.bin" 0x01000 $(FW_BASE)/user1.bin \
	  $(ET_BLANK) $(SDK_BASE)/bin/blank.bin`

Is it ok? There's a note in the SDK report that says esp_init_data_default.bin should be written on flash at least once.

Any clue?

tve, 2016-09-29:
Yes, it does build, see https://travis-ci.org/jeelabs/esp-link
I briefly had that problem too, but I don't remember how I got rid of it. I think it was a make clean.
Thanks for pointing out the Makefile flash target, I overlooked that. I use wiflash. Yes you are correct that init data should be written once.

katast, 2016-09-29:
The only changes I made was add `user_rf_cal_sector_set()` in the `main`, and fix the lack of declaration for the var `realtime_stamp` in syslog.
After that, this is what I get

**// Output of the making of**

xxxxx@xxxxx:~/xxxxx/esp-link$ make clean;make
Makefile:73: Using SDK from /xxxxx/xxxxx/xxxxx/esp-open-sdk_2.0.0/sdk
make[1]: Entering directory '/xxxxx/xxxxx/xxxxx/esp-link/espfs/mkespfsimage'
rm -f mkespfsimage main.o
make[1]: Leaving directory '/xxxxx/xxxxx/xxxxx/esp-link/espfs/mkespfsimage'
Makefile:73: Using SDK from /xxxxx/xxxxx/xxxxx/esp-open-sdk_2.0.0/sdk
VERSION: esp-link master - 2016-09-29 21:38:24 - development
CC mqtt/mqtt.c
CC mqtt/mqtt_msg.c
CC mqtt/pktbuf.c
CC mqtt/mqtt_cmd.c
CC rest/rest.c
CC syslog/syslog.c
CC espfs/espfs.c
CC httpd/httpdespfs.c
CC httpd/auth.c
CC httpd/base64.c
CC httpd/httpd.c
CC user/user_main.c
CC serial/crc16.c
CC serial/uart.c
CC serial/console.c
CC serial/slip.c
CC serial/serled.c
CC serial/serbridge.c
CC cmd/handlers.c
CC cmd/cmd.c
CC esp-link/heap.c
CC esp-link/cgimqtt.c
CC esp-link/log.c
CC esp-link/mqtt_client.c
CC esp-link/cgiflash.c
CC esp-link/config.c
CC esp-link/cgiwifi.c
CC esp-link/cgitcp.c
CC esp-link/cgiservices.c
CC esp-link/cgioptiboot.c
CC esp-link/cgi.c
CC esp-link/task.c
CC esp-link/main.c
CC esp-link/cgipins.c
CC esp-link/umm_malloc.c
CC esp-link/status.c
make[1]: Entering directory '/xxxxx/xxxxx/xxxxx/esp-link/espfs/mkespfsimage'
cc -I.. -std=gnu99 -DESPFS_GZIP   -c -o main.o main.c
cc -o mkespfsimage main.o -lz
make[1]: Leaving directory '/xxxxx/xxxxx/xxxxx/esp-link/espfs/mkespfsimage'
Compression assets with htmlcompressor. This may take a while...
Compression assets with yui-compressor. This may take a while...
style.css        ( 31%, gzip, 2276 bytes)
mqtt.js          ( 41%, gzip,  736 bytes)
console.html     ( 35%, gzip, 1188 bytes)
favicon.ico      (100%, none,  876 bytes)
home.html        ( 37%, gzip, 1972 bytes)
log.html         ( 39%, gzip,  908 bytes)
mqtt.html        ( 36%, gzip, 1200 bytes)
ui.js            ( 37%, gzip, 2804 bytes)
services.html    ( 34%, gzip, 1164 bytes)
services.js      ( 42%, gzip,  804 bytes)
console.js       ( 44%, gzip,  964 bytes)
pure.css         ( 22%, gzip, 3820 bytes)
wifi/wifiAp.js   ( 45%, gzip,  892 bytes)
wifi/wifiSta.js  ( 36%, gzip, 1664 bytes)
wifi/wifiSta.html ( 35%, gzip, 1148 bytes)
wifi/icons.png   (100%, none,  428 bytes)
wifi/wifiAp.html ( 34%, gzip, 1020 bytes)
24 -rw-rw-r-- 1 xxxxx xxxxx 24384 Sep 29 21:38 build/espfs.img
AR build/httpd_app.a
LD build/httpd.user1.out
/xxxxx/xxxxx/xxxxx/esp-open-sdk_2.0.0/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: build/httpd.user1.out section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
Makefile:348: recipe for target 'build/httpd.user1.out' failed
make: *** [build/httpd.user1.out] Error 1

**// After changing the ld values of iram from 0x8000 to 0x9000**

Makefile:73: Using SDK from /xxxxx/xxxxx/xxxxx/esp-open-sdk_2.0.0/sdk
VERSION: esp-link master - 2016-09-29 21:32:24 - development
ls -ls eagle*bin
  4 -rwxrwxr-x 1 xxxxx xxxxx   3081 Sep 29 21:32 eagle.app.v6.data.bin
260 -rwxrwxr-x 1 xxxxx xxxxx 264348 Sep 29 21:32 eagle.app.v6.irom0text.bin
 16 -rwxrwxr-x 1 xxxxx xxxxx  14524 Sep 29 21:32 eagle.app.v6.rodata.bin
 36 -rwxrwxr-x 1 xxxxx xxxxx  34636 Sep 29 21:32 eagle.app.v6.text.bin
-1386374458
1386374457
** user1.bin uses 316660 bytes of 503808 available
LD build/httpd.user2.out
854168809
854168810

**// Mem usage of the output with iram increased from 0x8000 to 0x9000**

xxxxx@xxxxxx:~/xxxxxx/esp-link$ OBJDUMP=../esp-open-sdk_2.0.0/xtensa-lx106-elf/bin/xtensa-lx106-elf-objdump ./mem_usage.sh build/httpd.user1.out 81920
data: 3081
rodata: 14524
bss: 39104
total: 56709
free: 25211

I really don't have any clue :-(

katast, 2016-09-29:
Solved.

closed 2016-09-29 by katast

tve, 2016-09-29:
what was the solution? switching to the independent SDK as opposed to the one embedded in esp-open-sdk?

katast, 2016-09-30:
The main problem was umm_malloc, which I prefer to mem_manager so I have just polished functions and removed some code that I'm not going to use anymore. 

With the limited amount of ram we have, I guess at some point we have to remove the less relevant code from the project. Waiting for a better and configurable, really open source, SDK.