jeelabs/esp-link #195: make: *** No rule to make target '/ld/eagle.app.v6.new.1024.app1.ld', needed by 'build/eagle.esphttpd1.v6.ld'.  Stop.

gojimmypi, 2016-10-02:
Hi. Newbie here. Trying to compile on RPi. 
I already have xtensa-lx106-elf installed here (and confirmed it is in my path; working as I use to compile MicroPython successfully):

/home/pi/workspace/esp-open-sdk/xtensa-lx106-elf/bin:/usr/local/sbin

after downloading and attempting to compile like this:

git clone https://github.com/jeelabs/esp-link.git
cd esp-link
make

I get this result:

pi@raspberrypi:~/workspace/esp-link $ make
Makefile:74: Using SDK from
VERSION: esp-link v2.2.3-86-gdb8bba0 - 2016-10-02 00:05:08 - db8bba0
CC mqtt/mqtt.c
CC mqtt/mqtt_msg.c
CC mqtt/pktbuf.c
CC mqtt/mqtt_cmd.c
CC rest/rest.c
CC web-server/web-server.c
CC espfs/espfs.c
CC httpd/httpdespfs.c
CC httpd/auth.c
CC httpd/multipart.c
CC httpd/base64.c
CC httpd/httpd.c
CC user/user_main.c
CC serial/crc16.c
CC serial/uart.c
CC serial/console.c
CC serial/slip.c
CC serial/serled.c
CC serial/serbridge.c
CC cmd/handlers.c
CC cmd/cmd.c
CC esp-link/cgimqtt.c
CC esp-link/log.c
CC esp-link/mqtt_client.c
CC esp-link/cgiflash.c
CC esp-link/cgitcp.c
CC esp-link/cgiservices.c
CC esp-link/cgioptiboot.c
CC esp-link/cgiwebserversetup.c
CC esp-link/config.c
CC esp-link/task.c
CC esp-link/cgi.c
CC esp-link/main.c
CC esp-link/cgipins.c
CC esp-link/status.c
CC esp-link/cgiwifi.c
make[1]: Entering directory '/home/pi/workspace/esp-link/espfs/mkespfsimage'
gcc -I.. -std=gnu99 -DESPFS_GZIP   -c -o main.o main.c
gcc -o mkespfsimage main.o -lz
make[1]: Leaving directory '/home/pi/workspace/esp-link/espfs/mkespfsimage'
Compressing assets with htmlcompressor. This may take a while...
Compressing assets with yui-compressor. This may take a while...
pure.css         ( 22%, gzip, 3820 bytes)
flash.html       ( 48%, gzip,  560 bytes)
ui.js            ( 37%, gzip, 2804 bytes)
userpage.js      ( 34%, gzip, 1264 bytes)
home.html        ( 36%, gzip, 2020 bytes)
mqtt.js          ( 41%, gzip,  736 bytes)
console.js       ( 44%, gzip,  964 bytes)
web-server.html  ( 54%, gzip,  500 bytes)
flash.js         ( 48%, gzip,  396 bytes)
mqtt.html        ( 36%, gzip, 1200 bytes)
wifi/icons.png   (100%, none,  428 bytes)
wifi/wifiSta.html ( 35%, gzip, 1148 bytes)
wifi/wifiAp.html ( 34%, gzip, 1020 bytes)
wifi/wifiAp.js   ( 45%, gzip,  892 bytes)
wifi/wifiSta.js  ( 36%, gzip, 1664 bytes)
services.html    ( 34%, gzip, 1164 bytes)
log.html         ( 39%, gzip,  908 bytes)
favicon.ico      (100%, none,  876 bytes)
console.html     ( 31%, gzip, 1300 bytes)
style.css        ( 31%, gzip, 2276 bytes)
services.js      ( 42%, gzip,  804 bytes)
28 -rw-r--r-- 1 pi pi 27380 Oct  2 00:06 build/espfs.img
AR build/httpd_app.a
make: *** No rule to make target '/ld/eagle.app.v6.new.1024.app1.ld', needed by 'build/eagle.esphttpd1.v6.ld'.  Stop.

I have confirmed zlib1g-dev and  python-serial are installed.

any tips on the /ld.. target? I didn't find it in the makefile.

thanks in advance,

tve, 2016-10-02:
You need to point to the Espressif SDK 2.0.0 with patch 1 applied. This line needs to print the location correctly: `Makefile:74: Using SDK from`. The error you're getting indicates that you are using the SDK bundled with esp-open-sdk, which isn't gonna work.

tve, 2016-10-02:
BTW, unless you want to hack on esp-link, you're better off getting a release binary from the github releases tab...

closed 2016-10-02 by tve