jeelabs/esp-link #197: ESP-01 after flashing esp-link 2.2.3 uses old credentials

rudi48, 2016-10-08:
I have tried to use esp-link version 2.2.3 on an ESP-01.
The UARTtoSerial module has an external 3.3 V regulator for up to 500 mA.

**Programming on Mac OS**:
$ ./esptool.py --port /dev/tty.SLAB_USBtoUART  write_flash -fs 4m -ff 40m 0x00000 boot_v1.5.bin 0x1000 user1.bin 0x7E000 blank.bin
esptool.py v1.0.2-dev
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0000
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (90.5 kbit/s)...
Writing 307200 @ 0x1000... 307200 (100 %)
Wrote 307200 bytes at 0x1000 in 26.6 seconds (92.3 kbit/s)...
Writing 4096 @ 0x7e000... 4096 (100 %)
Wrote 4096 bytes at 0x7e000 in 0.4 seconds (90.6 kbit/s)...
Leaving...

**Logging after a RESET**
   299> ** esp-link v2.2.3 - 2016-06-21 21:58:48 - 1bcdc62
   299> Flash config restore *FAILED*
   300> CONN led=0
   302> Wifi init, mode=AP+STA
   304> Wifi uses DHCP, hostname=esp-link
   308> "ip": "0.0.0.0"
   310> "netmask": "0.0.0.0"
   313> "gateway": "0.0.0.0"
   315> "hostname": "esp-link"
   318> sleep enable,type: 2
   321> Httpd init, conn=0x3fff2a6c
   324> Serbridge pins: reset=12 isp=13 swap=0
   337> Reset cause: 6=external
   337> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   337> Flash map 512KB:256/256, manuf 0xC8 chip 0x4013
   342> ** esp-link v2.2.3 - 2016-06-21 21:58:48 - 1bcdc62: ready, heap=25112
   349> initializing user application
   352> Waiting for work to do...
   355> mode : sta(18:fe:34:9d:4d:95) + softAP(1a:fe:34:9d:4d:95)
   361> add if0
   363> add if1
   364> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   372> bcn 100
  2496> scandone
  2497> no FRITZ-7170R found, reconnect after 1s
  2497> Wifi disconnected from ssid FRITZ-7170R, reason no_ap_found (201)
  2597> reconnect
  4721> scandone
  4721> no FRITZ-7170R found, reconnect after 1s
  4721> Wifi disconnected from ssid FRITZ-7170R, reason no_ap_found (201)
  4821> reconnect
repeating .....

It looks like, that **esp-link is using old credentials**, instead of switching to AP-mode
and serving a web page.

I do not understand the log line:
337> Reset cause: 6=external

What can I do, that esp-link starts in AP mode, to enter new credentals?

Regards, Rudolf

tve, 2016-10-08:
You log shows it running in STA+AP mode, so you should see its AP:
> 355> mode : sta(18:fe:34:9d:4d:95) + softAP(1a:fe:34:9d:4d:95)

rudi48, 2016-10-08:
Hello Thorsten,
Thank you very much for your quick help.
It looks like, that I had not enough patience.
After a while I yould see the Access Point **ESP_9d4D95** on my Android tablet.
I set in the web browser 192.168.4.1. Then I could enter my credentials (SSID, password).

The handling in the web browser is **very slow.** Once you are used to it, and after some retries, it worked to set the credentials.

Now I see on the **Terminal**:
808110> HTTP POST /wifi/scan: 200, 64ms, h=19792
808254> Starting a scan
808681> Wifi check: mode=AP+STA status=5
808681> Wifi got IP. Going into STA mode..
808682> station: 00:22:f4:92:c1:d0 leave, AID = 1
808684> rm 1
808686> bcn 0
808687> del if1
808688> pm open,type:2 0
808691> mode : sta(18:fe:34:9d:4d:95)
808785> Turning OFF uart log

The Telnet connection works.
One thing is still wondering me. After 5 minutes of inactivity the Telnet connectioned is closed:
Telnet client: **Connection closed by foreign host.**
I checked that 2 times. What is the reason about?

Thank you very much for your help.

Regards, Rudolf

tve, 2016-10-08:
Glad it worked. I don't know why it's slow on your tablet. It's all very simple javascript and ajax calls. Quite snappy here on an old android tablet.
The Telnet closes due to an inactivity timeout. It's a std feature of the Espressif SDK and I didn't change it. Given the very limited number of open connections that the chip can handle I think some form of timeout is necessary to avoid getting locked out due to stuck-open connections. E.g., you would just have to walk out of range with your table and you'd end up with a stuck-open connection. If the timeout is a problem I recommend you print a periodic status message in your sketch (or whatever is attached).

rudi48, 2016-10-08:
Hello Thorsten,
Thank you very much for the extensive answer. I can now understand the reason.
Unfortunately I did not found before any information about it in the documentation.
And I did not know, that Espressif already had such a timeout in their software.

I can live very good with it, it is just good to know.
Regards, Rudolf

rudi48, 2016-10-08:
Hello Thorsten,
I have documented on my wiki page 3 **Transparent Bridge** application programs, which are working.
http://www.rudiswiki.de/wiki9/WiFi2AT-ESP8266#esp-link
Regards, Rudolf

tve, 2016-10-12:
Thanks!
Reminder to self: add info about the telnet timeout to the docs

commit 5a037a2c (2016-10-23): document telnet timeout, closes #197
closed 2016-10-23 by tve