jeelabs/esp-link #198: Arduino Due Support

sebgiles, 2016-10-11:
Is it currently possible to flash a sam3x chip (arduino due)? Readme hints that programming various arm processors is possible, however the command bossac, used to flash the chip, does not allow users to specify a network address in place of a serial port

tve, 2016-10-12:
I'm not familiar with the sam3x series. Most arm chips need a combination of boot0 and reset to get into serial programming mode. In general, esp-link can issue the sequence and I'm happy to add a new one given the necessary information. WRT specifying a network address, a work-around it to either use a different tool or to use serial port forwarding.

rocketjosh, 2016-12-09:
Hey there - I'd be interested in this as well. Let me know if I can help with testing! Thank you for the excellent work.

steamboating, 2017-01-18:
Because my wood-walker.com robot on which i am using ESP-Link and ESP-client is based on the Arduino DUE I would be happy to test such features also - but sorry I am not a big help in programming something like this...

rocketjosh, 2017-03-02:
Here is a little more information about how the bootloader works on Arduino Due:

http://playground.arduino.cc/Bootloader/DueBootloaderExplained

tve, 2017-06-18:
closing 'cause nothing has moved here in a very long time.

closed 2017-06-18 by tve