jeelabs/esp-link #199: The serial status LED doesn't seem to work

yesyesuk, 2016-10-12:
I had mentioned this in the Gitter chat a while ago but haven't found a solution yet. It might be best to open an issue.

I just can't get the serial LED to light up at all when serial data is sent to/from the ESP. The WiFi LED works perfectly fine.

I have of course set the pin assignment in the Web UI to the pin my serial LED is connected to (GPIO14). 

I have designed a small PCB around the ESP-03 module that holds all the electronics needed to run the ESP-03 and a level shifter for the serial data line to to the ESP. The header on my module supplies the module with power, has serial TX and RX lines and the reset line for software upload to the Arduino. It plugs into my own home automation devices based on Arduino Minis. I have attached the schematics, mainly to show how the LEDs are connected (and photos to get an idea what the module looks like).

The ESP-03 has been flashed with esp-link 2.2.3.

Does anyone else use the serial LED and does it work for you?
Any idea why this doesn't work for me?

[ESP-03_Adapter - Schematic.pdf](https://github.com/jeelabs/esp-link/files/524153/ESP-03_Adapter.-.Schematic.pdf)
[image]
[image]

tve, 2016-10-12:
nice board! I'll have to look into the serial LED issue.

yesyesuk, 2016-10-12:
Thanks! I designed this PCB myself using DesignSpark PCB and had it manufactured in China.
It would be a great help for troubleshooting to get the serial LED working.
You may remember me mention this on the chat. After trying a few things to no avail I thought it might be best to open a "ticket" to track the issue. ;-)

closed 2016-10-23 by tve

yesyesuk, 2016-10-30:
I can confirm that 3.0 alpha0 fixes the serial LED issue. Thanks so much!

commit 3cb55b56 (2016-10-23): fix serial LED, fixes #199