jeelabs/esp-link #202: No AP is being created

Bavilo, 2016-10-19:
I have flashed the latest release 2.2.3.

After power cycling the device and letting GPIO0 float again, the device does not create an AP. Thus I can not connect to it.

Why am I having this issue?

Note: I have used blank, boot, and esp_init_data_default from the espressif 1.5.4 sdk

tve, 2016-10-19:
What is it saying on the uart1 (gpio2) debug log?
Did you wait ~ 1 minute?
Do you have an LED on gpio0 that shows you the blink pattern that indicates what it's doing?

nemi1, 2016-10-19:
tve - I am really looking forward to trying out ESP-link. I wrote my own audino code to cope with the quirks and unreliability of AT  firmware 0.9.2 but a direct TCP link sounds much easier on both PC and microprocessor.

I'm having a similar issue to "Bavilo" above with my 8mbit ESP-01 device. I will build a new programming board so I can spy on GPIO2 and report back.

I am using windows with ESP flash download tool v2.3
It has detected my device and reports 8Mbit.

This site suggests
boot_v1.X.bin 0x00000
user1.bin to 0x01000
esp_init_data_default.bin 0xFC000 
blank.bin 0xFE000

Is that correct?

note 1: I found a copy of "esp_init_data_default.bin" in SDK 2.0.0_16. The rest of the files are from the latest release here.

note 2: I have programmed AT firmware v0.9.2 using "esp8266_flasher.exe" to 0x0000 and that works fine so I think my programming hardware is ok.

Request /suggestion 1 - it would be really helpful to have a single .bin file release of ESP-link that could be programmed at location 0x0000 using these easy window flasher tools. 

Request /suggestion 2 - at boot up could esp link dump some status to the regular Tx port? If each status line was prefixed with some charters (e.g. "###" ) the external microprocessor could ignore or use the data to make decisions,

Request /suggestion 3  - does anyone sell esp-link pre-programmed ESP-01devices for those of us having programming troubles? :-)

Bavilo, 2016-10-20:
@nemi1 please report back if you find a way to get this to work on our device. Currently I am having no luck with this.

Bavilo, 2016-10-22:
I have it working now!

Turns out that my esp-01 was simply dead. I have ordered a new one and flashed esp-link with the v1.4 boot.bin without any problems. I haven't tried the v1.5 boot.bin because @tve said it was faulty. 

Thanks for the great work!

tve, 2016-10-23:
Perfect. Thanks for reporting back!

closed 2016-10-23 by tve

nemi1, 2016-10-23:
Bavilo
Exactly which files, To what address locations from, which release / download links to make a 8mbit esp-01 flash correctly? Many thanks.

Bavilo, 2016-10-23:
Sure thing!

I have downloaded the latest Espressif SDK version and used the esp_init_data_default.bin from it.
http://bbs.espressif.com/viewtopic.php?f=46&t=2451

Then I have downloaded the v2.1.7 esp-link firmware and used boot_v1.4(b1).bin from it.
https://github.com/jeelabs/esp-link/releases/tag/v2.1.7

Then I have downloaded the v2.2.3 esp-link firmware and used blank.bin and user1.bin from it.
https://github.com/jeelabs/esp-link/releases/tag/v2.2.3

To flash everything I have used the Flash Download Tool from Espressif:
https://espressif.com/sites/default/files/tools/flash_download_tools_v3.4.2_win.zip

Once you have all the needed files, set the Flash Download Tool up like so:
[image]

Worked like a charm! Please report back :)

nemi1, 2016-10-23:
Bavilo - that is great, thank you :-)
Following your instructions I have a good flash and able to connect on 192.168.1.4 afterwards.
It looks like I had the addresses correct and it was the use of boot v1.5 that caused me the issues.

Now I just need to figure out how to make it join my home AP.  the wifi-station page is listing no AP's in it's scan and when I manually enter the AP name and PW I get error "Error switching network: Can't associate to an AP en SoftAP mode".

Bavilo, 2016-10-23:
I think you need to switch from AP Mode to STA Mode. Its somewhere in the gui

nemi1, 2016-10-23:
all working :-)

I fired up 2 Putty sessions. one telnet to the IP address of the ESP and the other to the com port of the USB TTL serial adapter it is still attached from for programming.

Characters are flowing both ways as I type. A very impressive piece of software (firmware). :-) 

Arguably this should have been the default mode / feature the ESP-01 shipped with when the launched (coding for the 0.9.x AT firmware on the arduino side was overly complex).

tve - this is great, thank you :-)

hiperz, 2020-05-18:
[image]

Thanks a lot!
I use serial connection through Aduino UNO for flashing.
In my case, SPI mode "DOUT" only makes it work.