jeelabs/esp-link #204: Initial build problem

dannybackx, 2016-10-23:
Hi,
I have a Arduino Uno Wifi and have my app mostly working on it but its limitations lead me here. I may toss the Uno and replace it with e.g. Arduino Mega + ESP (Nodemcu board).

I have the esp-link build almost complete, but now this error occurs :
`acer: {140} make
Makefile:78: Using SDK from /home/danny/src/esplink/ESP8266_NONOS_SDK
VERSION: esp-link_custom_version
LD build/httpd.user1.out
@ /home/danny/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc -L/home/danny/src/esplink/ESP8266_NONOS_SDK/lib -Tbuild/eagle.esphttpd1.v6.ld -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--gc-sections -Wl,--start-group -lc -lgcc -lphy -lpp -lnet80211 -lwpa -lmain -llwip -lcrypto build/httpd_app.a -Wl,--end-group -o build/httpd.user1.out
/home/danny/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: build/httpd.user1.out section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
Makefile:361: recipe for target 'build/httpd.user1.out' failed
make: *** [build/httpd.user1.out] Error 1
`
This is with esp8266_nonos_sdk_v2.0.0_16_08_10.zip .

What does this mean, how can I fix ?

Thanks,

  Danny

tve, 2016-10-23:
The error says that you end up with too much code in the 32KB on-chip instruction RAM. I've seen this happen but don't quite know the cause. Make sure to pull the latest code and run a make clean. You might also pull the just-baked v3.0.alpha0 release. Yet another option is to use the docker image to do the build (or to compare against your build).

dannybackx, 2016-10-23:
That's odd. This is the software as I've pulled it from github a few hours ago.

tve, 2016-10-23:
I've been busy ;-)

dannybackx, 2016-10-23:
I've just pulled 3.0alpha0 as you said, and applied just the Makefile changes I needed.
The result is the same.
Not sure if this matters, one of the last things built is : 
`-rw-rw-r-- 1 danny danny 27384 Oct 23 09:54 build/espfs.img`

dannybackx, 2016-10-23:
I tried v2.2.3 too, and it looks like I have the same problems as described in issue https://github.com/jeelabs/esp-link/issues/194 .
What should I do ?

tve, 2016-10-23:
You can't link v2.2.3 with SDK 2.0, they're not compatible. Each esp-link version states which SDK version is requires. Espressif changes something in the SDK at each round...

Dunno how to help you, you have to get to a state that compiles and then apply your changes, etc. You could try the docker container. Or you could try the esp-open-sdk and/or SDK that I use in the travis-ci build: https://github.com/jeelabs/esp-link/blob/master/.travis.yml#L6-L7

I know it's frustrating, ease of build-env or ease of flashing are not hallmarks of the esp8266 :-(

dannybackx, 2016-10-23:
I realized I can download images too, this is a good way for me to get started.
I'll fiddle with that a bit and see whether this helps me further than the stock Uno Wifi environment.

Thanks !

closed 2016-10-23 by dannybackx