jeelabs/esp-link #207: Basic question about flashing the firmware

FernandoGarcia, 2016-10-30:
Hi!

Sorry for this basic question but I can't find the answer in anywhere.

In [this instructions](https://github.com/jeelabs/esp-link/blob/master/FLASHING.md) say:

Download the latest release or use the user1.bin and user2.bin files that are produced by the build process. You will need to flash the bootloader, the firmware, blank wifi settings, and init data. Detailed instructions are provided in the release notes.
.
.
.
The short version for the serial flashing is:

    flash boot_v1.X.bin from the official SDK or from the release tgz to 0x00000
    flash blank.bin from the official SDK or from the tgz to 0x3FE000
    flash esp_init_data_default.bin from the official SDK or from the tgz to 0x3FC000
    flash user1.bin to 0x01000
    be sure to use the commandline flags when flashing the bootloader to set the correct flash size
    some of the addresses vary with flash chip size

What's the address for user2.bin when using a ESP-12 (nodeMCU V3) or it's alternative to user1.bin?

Best regards.

blackadmin, 2016-11-01:
I am using ESP12F 4MB and address for user2.bin I use is 
**0x81000** firmware/user2.bin

FernandoGarcia, 2016-11-01:
Hi!
@blackadmin Thanks for your answer!
@tve can you confirm this info and add it to documentation?
Best regards.

hetii, 2016-11-01:
Well ...
I flash my module by: `esptool.py --port /dev/ttyUSB0 --baud 230400 write_flash -fs 32m -ff 80m \
    0x00000 boot_v1.5.bin 0x1000 user1.bin \
    0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin`
and  it boot fine, 
so for what is user2.bin ?

BTW I also try compile tags/v2.2.3
but it end on: `make: *** No rule to make target '/ld/eagle.app.v6.new.1024.app1.ld', needed by 'build/eagle.esphttpd1.v6.ld'.  Stop.`

Any success to build it from source ?

blackadmin, 2016-11-01:
Might this help https://github.com/jeelabs/esp-link/issues/155

tve, 2016-11-01:
user2.bin is only needed for over-the-air updates. The serial flashing does not need user2.bin.

closed 2016-11-01 by tve

blackadmin, 2016-11-01:
@tve Yes you are right, but there was a scenario with me when esp was crashing on boot and the crash was due to code in user2.bin, in that case user2.bin has to be flashed serially.

commit 0df16c98 (2016-11-01): fix flashing instructions, fixes #207