jeelabs/esp-link #209: Full STK500v2.

hetii, 2016-10-30:
Till now esp-link require to have boot-loader already in AVRs.
I have some success to port stk500v2 for esp-open-rtos and I`m able to flash avrs without boot-loader by ESP8266.

If you think that it could be useful to have it, I can port it as well for esp-link and create PR.
Please let me know what you think about this idea.

ramarro123, 2016-11-20:
hi i would like to take a look to your work and see if it's possible to port it to esp8266-arduino framework. can you point me on where is your code?

tve, 2016-11-30:
There definitely is demand for stk500v2 support. A PR would be awesome, failing that, could you provide a link to your code?

closed 2016-12-11 by tve